ARMmbed/mbed-os #6699: Change "macros" in module.json?

pingdan32, 2018-04-21:
### Description
- Type: Question
- Related issue: `#abc`
- Priority: mbed-os 5.8.2
-----------------------------------------------------------------

## Question
I'm sorry to post a question via `issue` because I can't find some content elsewhere. 

How can I modify the value of macros defined in `module.json`? For example, the default value of `NRF_SDH_BLE_GATT_MAX_MTU_SIZE` (defined in `TARGET_NORDIC/TARGET_NRF5x/TARGET_SDK_14_2/TARGET_SOFTDEVICE_S132/module.json`) is 23, and I want to change it to another value in `mbed_app.json`.

Thanks!

0xc0170, 2018-04-23:
`target.macros_remove` to remove the macro and add your new one, does this work for you?

pingdan32, 2018-04-23:
@0xc0170 
Thanks for your help, but this does not seem to work. 

`NRF_SDH_BLE_GATT_MAX_MTU_SIZE=23` is defined in the `macros` of `targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_SDK_14_2/TARGET_SOFTDEVICE_S132/mbed_lib.json`.I want to change to NRF_SDK_BLE_GATT_MAX_MTU_SIZE=92 in mbed_app.json, but I didn't find a practical implementation. 

My way is as follows(mbed_app.json):
`{
    "config": {
        "main-stack-size": 4096,
        "thread-stack-size": 4096
    },
    "target_overrides": {
        "*": {
            "target.console-uart-flow-control": null,
            "target.macros_remove": ["NRF_SDH_BLE_GATT_MAX_MTU_SIZE"],
            "target.macros_add": ["NRF_SDH_BLE_GATT_MAX_MTU_SIZE=92"]
        }
    }
}`
But it doesn't work. I enter the following command `mbed compile --config` in the terminal and wait until the following result :

Macros
------
NRF_BLE_GATT_BLE_OBSERVER_PRIO=1
NRF_SD_BLE_API_VERSION=5
NRF_SDH_CLOCK_LF_XTAL_ACCURACY=7
NRF_SDH_BLE_VS_UUID_COUNT=4
NRF_SDH_CLOCK_LF_SRC=1
NRF_SDH_CLOCK_LF_RC_CTIV=0
BLE_CONN_STATE_BLE_OBSERVER_PRIO=0
NRF_SDH_BLE_GATT_MAX_MTU_SIZE=23
others...

0xc0170, 2018-04-23:
I could not find it on master, what mbed OS version are you using? I cant find it even on 5.8 or earlier versions like 5.2, so must be older?

To extend my previous answer, I suspected it is target config but this rather library configuration. You can also review `mbed_config.h` (docs https://docs.mbed.com/docs/mbed-os-handbook/en/latest/advanced/config_system/). 

targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_SDK_14_2/TARGET_SOFTDEVICE_S132/mbed_lib.json - can you share how that macro is defined?

0xc0170, 2018-04-24:
It should be `softdevice` name there as it's the name of the config that set that macro.

@marcuschangarm Can you help ?

marcuschangarm, 2018-04-24:
It looks like the config system doesn't support removing a macro and then adding it again with a different value. 

@theotherjimmy can you confirm if that is the case?

theotherjimmy, 2018-04-24:
It should....

0xc0170, 2018-04-24:
@marcuschangarm This is unfortunate, thus any softdevice config that is there can't be changed.

theotherjimmy, 2018-04-24:
@0xc0170 You could "upgrade" the softdevice macros to config values, then it would work.

marcuschangarm, 2018-04-24:
> library macros are not configuration values, and cannot be modified as a config value. target macros are modifiable though.

Is that by design?

theotherjimmy, 2018-04-24:
I think so? I'm pretty sure it's always been like that...

marcuschangarm, 2018-04-24:
😄 

As a user writing mbed_app.json, it seems like an odd difference.

theotherjimmy, 2018-04-24:
Yeah, Agreed.

pingdan32, 2018-04-25:
Yeah, it is the best way to modify the library macros definition as a configuration so far.

However, there are many configurations associated with other contents. For example, modifying the GATT MTU value of SOFTDEVICE may require modifying APP_RAM_START. So, I suggest that it is best to be implemented by the Python compilation system (in `tools`) so that it can modify the implementation of nRF52832.ld(icf/sct) at the same time. But it may require more work.

It may be necessary to implement a complete set of calculation methods. When the user modifies a value in mbed_app.json, the compiling system should calculate all relevant content based on this calculation method and automatically modify it to the corresponding result. For example, if the user has modified CENTRAL_LINK_COUNT, the build system should be able to calculate the required APP_RAM_START or other. I think this is the most friendly implementation.

theotherjimmy, 2018-04-25:
@pingdan32 

> I suggest that it is best to be implemented by the Python compilation system (in tools) so that it can modify the implementation of nRF52832.ld(icf/sct) at the same time. But it may require more work.

Agreed, that requires quite a bit more work, and agreed, that would be nice.

>It may be necessary to implement a complete set of calculation methods.

I would like to see a current implementation of those calculation methods first. I don't mean that you have to provide them; I'll find them if I get time to design and implement these changes.

pingdan32, 2018-04-26:
@theotherjimmy 

> I would like to see a current implementation of those calculation methods first. I don't mean that you have to provide them; I'll find them if I get time to design and implement these changes.

Agreed ! Great!

0xc0170, 2018-05-22:
@ARMmbed/mbed-os-tools Please review. Is there any outstanding issue here ?

theotherjimmy, 2018-05-22:
There is, and it's tagged correctly.

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCORE-337*

closed 2019-10-31 by ciarmcom