ARMmbed/mbed-os #6706: NVStore call crashes when compiled with ARMCC

romkuz01, 2018-04-23:
Use the following main.cpp to reproduce the crash:

```
#include "nvstore.h"

const uint8_t global_buffer[91] = {0}; // doesn't crash without "const"

int main()
{
     NVStore &nvstore = NVStore::get_instance();

     printf("Trying to reproduce NVStore crash...\n");
     nvstore.set(10, sizeof(global_buffer), global_buffer);
     printf("NVStore didn't crash...\n");
}
```

Compile and run with:
`mbed compile -t ARM -m K64F --profile debug -f`

The failure was detected with ARMCC version: 
`ARM Compiler 5.06 update 3 (build 300)`

Some observations:

- Crashes only with --profile set to "debug" or "release", doesn't crash with "develop" 
- Passes when "const" removed from array definition
- Passes if array is defined as uint32_t (change in data alignment?)

0xc0170, 2018-04-23:
@davidsaada Please review

davidsaada, 2018-05-10:
Resolved by PR #6864.

closed 2018-08-02 by cmonr