ARMmbed/mbed-os #6716: Wrong #define macro name in tls unit test

Alex-EEE, 2018-04-23:
### Description
- Type: Bug 

In the unit test for mbedtls ( mbed-os\TESTS\mbedtls\selftest\main.cpp ), the test refers to the mbed tls config file by the wrong name.  It is currently coded as:
#if !defined(MBEDTLS_CONFIG_FILE)
#include "mbedtls/config.h"
#else
#include MBEDTLS_CONFIG_FILE
#endif

****** It should be MBEDTLS_USER_CONFIG_FILE  *************

Greentea currently fails like so:

* K64F::IAR::MBED-MBED-OS-TESTS-MBEDTLS-SELFTEST
        Building project selftest (K64F, IAR)
        Scan: IAR
        Scan: selftest
        Scan: FEATURE_LWIP
        Scan: FEATURE_COMMON_PAL
        Scan: FEATURE_STORAGE
        Compile [100.0%]: main.cpp
        [Warning] main.cpp@71,7: [Pe035]: #error directive: "MBEDTLS_SELF_TEST not enabled"
        [Error] main.cpp@74,30: [Pe1345]: an empty initializer is invalid for an array with unspecified bound
        [Error] main.cpp@97,0: [Pe289]: no instance of constructor "utest::v1::Specification::Specification" matches the argument list

**Target**
K64F

**Toolchain:**
IAR

0xc0170, 2018-04-24:
@hanno-arm @mazimkhan Please review

sbutcher-arm, 2018-10-01:
This isn't a bug and is by design, although I think it's confusing. Looking at our [documentation](https://os.mbed.com/docs/v5.10/apis/tls.html#configuring-mbed-tls-features), I can see it isn't clear, so I think we have something to fix here.

There are several different ways of configuring Mbed TLS - but aside from configurations in the JSON file, you can use the Mbed OS default platform configurations, and add to them with your own configuration file, by defining `MBEDTLS_USER_CONFIG_FILE` to point to your own configuration file, or you can overwrite all configurations by defining `MBEDTLS_CONFIG_FILE` and making it point to your own configuration file which replaces all others.

Internally, all Mbed TLS library files include `MBEDTLS_CONFIG_FILE` if it's defined, else they include the platform default configuration file `config.h`, which in turn includes `MBEDTLS_USER_CONFIG_FILE` if it's defined.

In the instance above, if `MBEDTLS_CONFIG_FILE` is not defined, `config.h` must be included, which will include `MBEDTLS_USER_CONFIG_FILE`. If `MBEDTLS_CONFIG_FILE` is defined, then `MBEDTLS_USER_CONFIG_FILE` will not be included by design.

I don't know why `MBEDTLS_SELF_TEST` wasn't defined, but there was likely a misconfiguration of which config file to use.

I hope that makes sense! I suggest we close this PR and reject the bug - unless there's something I've misunderstood.

Alex-EEE, 2018-10-02:
Ok, thanks, I'll take a look at that documentation.

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTSSL-2328*

hanno-arm, 2018-10-17:
@sbutcher-arm What you wrote should be part of the public documentation on how to configure [Mbed TLS for use within] Mbed OS. Is it? If not, @adbridge where would be the right place for it?

adbridge, 2018-10-18:
> @sbutcher-arm What you wrote should be part of the public documentation on how to configure [Mbed TLS for use within] Mbed OS. Is it? If not, @adbridge where would be the right place for it?

@AnotherButler Any thoughts on where we could include something like this in the mbed-os docs?

closed 2019-06-16 by ciarmcom