ARMmbed/mbed-os #6722: TARGET_STM/gpio_irq_api.c does not need to guess which port generated the interrupt

michaelVisimid, 2018-04-24:
### Description
- Type: Enhancement (bug if you are feeling contrarian)

- Priority: Minor

---------------------------------------------------------------
## Bug

**Target**
All STM devices

**Toolchain:**
GCC_ARM|ARM|IAR

----------------------------------------------------------------
## Enhancement

There seems to be some misconceptions about STM32 devices and more than one pin using the interrupt lines. The microcontroller will provide both which pin and which port generated the interrupt request.

**Suggested enhancement**

Referencing STM RM0090 STM32F405/415, STM32F407/417, STM32F427/437 and
STM32F429/439 advanced ARM®-based 32-bit MCUs Section 9.3.3 - 9.3.6, though this information is valid regardless of the Fx or Lx part number, SYSCFG_EXTICRx register has the information as to which port generated.

I recommend in the handle_interrupt_in function to use the LL macro LL_SYSCFG_GetEXTISource to determine which port has generated the interrupt. Line 86 would be a good place for this change. Then the pins can be converted to the mbed format and passed to the irq_handler functions.

Is there a reason to store the pin mask in the gpio_channel struct? Is this faster than getting the same data from the microcontroller's registers? Seems a trivial amount of memory could be saved by getting rid of the pin mask and switching the pin and port arrays to mbed pin types.

**Pros**

Allows more than one exti line to be used. That is to say both PA_0 and PB_0 can have external interrupts. They're multiplexed and not OR'd after all. There is no reason to not just ask the microcontroller which port generated the interrupt via checking the multiplexor's control bits.

Relatively few changes. 

**Cons**

I realized this fix in a project I started based on some STM32 cube example code which has the same undesirable behavior. That is what led me to check and see if the mBed code base has this same quirk and / or bug. This means someone else would have to make the changes as I am pretty swamped right now.

It will require a bit of thought and likely some discussion to determine the best way to handle the changes that will need to be made so that each pin on each port can have its own call back.

0xc0170, 2018-04-24:
@ARMmbed/team-st-mcd Please review

LMESTM, 2018-04-24:
@michaelVisimid @0xc0170  Thanks for sharing the thought and detailed explanation. We'd welcome a Pull Request as well as we're a bit swamped as well now. This will require careful change a good non-regression testing. But it's now added to the TO-DO list ...

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTPART-6055*

bcostm, 2018-10-09:
I think this issue can be closed as it has been fixed with PR #8037

closed 2018-10-09 by 0xc0170