ARMmbed/mbed-os #6753: MAX32620FTHR not recognized in online compiler

yennster, 2018-04-26:
## Bug

**Target**
MAX32620FTHR

**Toolchain:**
Online Compiler

**mbed-os sha:**
Revision: 4127:f9ee4e8

**Expected behavior**
The online compiler recognizes the MAX32620FTHR target and compile's the program.

**Actual behavior**
Error: Target "MAX32620FTHR" is not recognized

**Steps to reproduce**
1. Add the board to your online compiler: https://os.mbed.com/platforms/MAX32620FTHR/
1. Import Blinky into the online compiler with the latest version of Mbed OS: https://os.mbed.com/compiler/#import:/teams/mbed-os-examples/code/mbed-os-example-blinky/
1. Compile, see not recognized error

The board is recognized by Mbed CLI however and compiles blinky for the target with GCC_ARM.

bentcooke, 2018-04-26:
This target was just released in v5.8.3.  The mbed os lib of blinky has not yet been updated to point to the latest lib.
Updating the mbed-os lib in the online project will fix it.

yennster, 2018-04-27:
@bentcooke Okay, I'll send a bug to the web team then

closed 2018-04-27 by yennster