ARMmbed/mbed-os #6754: Tests failing on EFM32GG_STK3700

bulislaw, 2018-04-26:
### Description
- Type: Bug 
- Priority:  Major
- Mbed OS: `675528b6c`

CC: @ARMmbed/team-silabs 

---------------------------------------------------------------
## Bug

**Target**
EFM32GG_STK3700

**Toolchain:**
GCC_ARM

**Expected behavior**
All tests pass.

**Actual behavior**
Fails:
```
features-nvstore-tests-nvstore-functionality
tests-mbed_drivers-timerevent
tests-mbed_platform-system_reset
```

**Steps to reproduce**
`mbed test`

stevew817, 2018-04-26:
Are these new tests? Do they fail on our other platforms?

bulislaw, 2018-04-26:
They are newish I guess, came in last month or two. They are passing on other boards in our CI.

0xc0170, 2018-04-27:
cc @studavekar @kegilbert (for CI and this target)

0xc0170, 2018-05-22:
> Are these new tests? Do they fail on our other platforms?

@ARMmbed/mbed-os-test Please update this ticket

stevew817, 2018-06-27:
@ARMmbed/mbed-os-test is this still an issue?

0xc0170, 2018-06-27:
> @ARMmbed/mbed-os-test is this still an issue?

@stevew817 Can you retest? I can't locate GG board and checked latest CI results (GG was not there)

cmonr, 2018-08-02:
@studavekar @OPpuolitaival Was there a Jira issue in regards to this? Iirc, the GG use to be in CI, but was removed for stability reasons.

studavekar, 2018-08-02:
@cmonr I don't recollect having this board mbed-os CI.

cmonr, 2018-08-07:
Going to locally test to see if this is still an issue.

cmonr, 2018-08-10:
From what I can tell, the EFM32GG_STK3700 consistently fails the "lp ticker re-init after free test" from the test-mbed_hal-common_tickers test.

There are one or two other tests that it fails with a timeout, but not consistently. Since CI automatically reattempts tests up to three times, I don't see those as a problem (in the context of this issue).

cmonr, 2018-08-13:
@ARMmbed/team-silabs Fyi. ^^^

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTPART-6201*

adbridge, 2020-10-13:
We've updated our automation, I will fix the requirements .

adbridge, 2021-06-07:
As there has been no update on this for over 2 years I am closing it. Please re-open if this is still a valid issue.