ARMmbed/mbed-os #6755: [exporter] cmake-gcc-arm doesn't include FEATURE_LWIP

yennster, 2018-04-26:
## Bug

**Target**
LPC1768

**Toolchain:**
Online Compiler, CMake-GCC-ARM

**mbed-os sha:**
Revision: 4127:f9ee4e8

**Expected behavior**
Exporting mbed-os-example-blinky with LPC1768 and CMake-GCC-ARM does not include references to the FEATURE_LWIP files in CMakeLists.txt under `ADD_EXECUTABLE(...` or grabs the FEATURE_LWIP folder in the mbed-os/features directory

**Actual behavior**
The exporter generates a .zip file for CMake-GCC-ARM but the CMakeLists.txt includes references to the FEATURE_LWIP directory which causes cmake build errors because the FEATURES_LWIP folder does not exist within the mbed-os/features directory

**Steps to reproduce**
1. Add the board to your online compiler: https://os.mbed.com/platforms/mbed-LPC1768/
1. Import Blinky into the online compiler with the latest version of Mbed OS: https://os.mbed.com/compiler/#import:/teams/mbed-os-examples/code/mbed-os-example-blinky/
1. Right click program name, "Export Program...", select LPC1768 and CMake-GCC-ARM, download .zip file, extract, see mbed-os/features folder and CMakeLists.txt file

Question for reference: https://os.mbed.com/questions/81034/How-can-I-use-mbed-1768-on-CLion/#answer14504

0xc0170, 2018-04-27:
I tried locally offline latest master and also 5.8.3 tag checked out and cmake gcc arm, I can see lwip there in executables. 

@theotherjimmy Please review

theotherjimmy, 2018-04-27:
@yennster The expected behavior is that it contains references and code for LWIP.

yennster, 2018-04-27:
@theotherjimmy Yep, sorry. I worded that badly

yennster, 2018-04-27:
@0xc0170 @theotherjimmy The issue is not that LWIP is included in executables in the CMakeLists.txt, but rather that the files are not showing up in mbed-os/features for LWIP

theotherjimmy, 2018-04-27:
@yennster Yep. I'm testing a fix right now.

cmonr, 2018-08-02:
@yennster Could you confirm that this issue has been fixed?

yennster, 2018-08-13:
@cmonr This issue is resolved, but unfortunately there's now a similar issue to this but with `nanostack` 😞I'll open up a new issue

closed 2018-08-13 by yennster