ARMmbed/mbed-os #6759: EventQueue Background Example

AGlass0fMilk, 2018-04-26:
### Description
- Type: Question
------------------------------------
## Question

How does one effectively use the EventQueue.background method? What does the passed in update function have to handle and what is the significance of the int argument?

An official example of this functionality would be helpful!

mattbrown015, 2018-04-27:
Coincidentally I was looking for an example of 'EventQueue.background' yesterday.  I too would like to see an official example.

geky, 2018-04-27:
Unfortunately I don't have the time to update docs right now, but I can try to explain it here and schedule time with @AnotherButler to get it into the online docs.

Let me know if this helps! Or if you guys are left with questions

---

So, event queue backgrounding!

Well, first, what does an event queue do? It moves tasks from one **execution context** to another **execution context**.

In the embedded space, we usually use the event queue to move irqs from high priority interrupt context to a low priority threaded context:
``` cpp
EventQueue equeue;

// example function
void print_callback() {
    printf("hi!\n");
}

// high priority interrupt context
void irq() {
    // we use the "call" method to pass the callback_function to a different execution context
    equeue.call(print_callback);
}

// low priority threaded context
int main() {
    // hardware timer, can only execute code in interrupt context
    Timer timer;
    timer.attach_ms(irq, 100);

    // dispatch events in our current context
    equeue.dispatch();
}
```

In this example, we assign the execution context for evens by calling the `dispatch` function. `dispatch` runs the event queue in the current execution context, in this case the low priority thread.

But `dispatch` has a downside. It consumes the whole thread! While we're running `dispatch` we can't execute anything else.

We can pass a timeout to `dispatch`, this would let us get our execution context back for periodic events:
``` cpp
// low priority threaded context
int main() {
    // hardware timer, can only execute code in interrupt context
    Timer timer;
    timer.attach_ms(irq, 100);

    while (true) {
        // dispatch events in our current context for 100 ms
        equeue.dispatch(100);

        // super fancy status update system
        led = !led;
    }
}
```

And we could even pass 0 ms to execute any pending events and return immediately, letting us spend all of our time blinking our LED. 

But that's still rather limiting. What if we want to do stuff in our thread, execute events, _and_ sleep?

What would this look like? We would need to be able to get how to long to sleep from the event queue, but since the event queue could be updated from any IRQ the time to sleep could change whenever it wants. So we also need a way to be notified when a new sleep time has been decided...

This is where the event queue `background` function comes in!

``` cpp
/** Background an event queue onto a single-shot timer-interrupt
 *
 *  When updated, the event queue will call the provided update function
[... 117 lines cut ...]
        // attach our sonar FSM to the provided event queue, defaulting to the global mbed event queue
        _equeue.chain(parent);
    }

    EventQueue _equeue(SONAR_EVENTS * EVENTS_EVENT_SIZE);
}
```

Normally, when you pass around an event queue, you need to track how many events could be allocated at once. But if event queues are chained together, the parent no longer needs to care about the child's quantity of events, as long as it gives the child the hook to dispatch its own event queue (1 event for the child's chain call).

In this way we can pass the decision of execution context entirely up to our caller:
``` cpp
// create three sonars using our event queue
EventQueue queue(3 * EVENTS_EVENT_SIZE);
Sonar s1(&queue)
Sonar s2(&queue);
Sonar s3(&queue);

int main() {
    // dispatch all of our sonars at once
    queue.dispatch();
}
```

**Note:** This is a form of controlling execution context **explicitly**. In mbed-os, you can also control execution context **implicitly** by creating different threads for each sonar. The decision for which one to use is left up to you.

mattbrown015, 2018-04-27:
Thanks. That's certainly given me plenty to be going on with! :-)

geky, 2018-04-29:
Glad I could help! It would be interesting to see more technical notes on the blog, I'll mention that to @janjongboom / @BlackstoneEngineering.

BlackstoneEngineering, 2018-04-29:
@geky , want to do an office hours on Event Queue?

kjbracey-arm, 2018-04-30:
I've never looked too hard at this - would it make sense to encourage this "chain" rather than "use directly" model for users of the shared event queues - putting it into example docs there?

Would allow us to maybe reduce the default event queue size for those, and also permits people who to use them who want to queue significant data (eg packet payloads).

geky, 2018-05-02:
I'm not sure if this approach is for everyone, it works if you know roughly how much memory the event queue for your module needs, and it's a good way to keep memory exhaustion limited to different modules. If you don't know, having multiple event queues may do more harm than good.

It'd be interesting to try it out for a code module and see if it has a net benefit.

For significant data, I think user allocated events is the correct solution, but I haven't been able to add support for user allocated events. It's probably the biggest missing feature of the equeue right now.

kjbracey-arm, 2018-05-02:
The chaining makes it more "lumpy", right? The thing dispatches everything from the chain as 1 top-level events?

User-allocated events is the biggest omission from my point of view. It's vital for robustness - being able to queue pre-allocated events with no risk of memory exhaustion. (Added this to Nanostack's event queue a couple of years ago to make it survive fuzz testing).

AGlass0fMilk, 2018-07-12:
If this hasn't been integrated into the official handbook already, it should be put _somewhere_ public. The explanation above was too good to be lost to the depths of closed GitHub issues.

kegilbert, 2018-07-12:
Thanks for the feedback! We're working on integrating it into our documentation, completely agree that Geky did a great write-up on this.

geky, 2018-07-12:
Here's the current pr for it:
https://github.com/ARMmbed/mbed-os-5-docs/pull/507

EmbedEngineer, 2018-08-28:
@geky I tried the background example with moving the dispatching of the queue into another thread and figured that it doesn't quite have the expected behaviour. The `sleep_callback` always gets a 0 as parameter after the first execution. So the controller never goes to sleep since the `sleep_time` is always 0 and the LED is just flickering instead of blinking. The "hi" is printed out correctly every second.

I used the following code:
```
EventQueue equeue;
DigitalOut led(LED0);
Thread thread1;

int sleep_time = -1;
Semaphore sleep_sema;

void sleep_callback(int new_sleep_time)
{
    // update sleep_time
    sleep_time = new_sleep_time;

    // go ahead and wake up thread so it can sleep for new sleep_time
    sleep_sema.release();
}

// example function
void print_callback()
{
    printf("hi!\n");
}

// high priority interrupt context
void irq()
{
    // we use the "call" method to pass the callback_function to a different execution context
    equeue.call(print_callback);
}

void threadContext()
{
    while (true)
    {
        // dispatch any pending events
        equeue.dispatch(0);

        // super duper fancy status update system
        led = !led;

        // go to sleep until next event
        sleep_sema.wait(sleep_time);
    }
}

void main(void)
{
    // hardware timer, can only execute code in interrupt context
    Ticker timer;
    timer.attach_us(irq, 1000000);

    // attach our sleep callback so we get sleep updates
    equeue.background(sleep_callback);

    thread1.start(threadContext);

    while (true)
    {
        sleep();
    }
}
```
I tried this code on an EFM32 Pearl Gecko DevBoard from Silicon Labs.
So my question is, am I doing something wrong here?

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCORE-340*

closed 2019-10-31 by ciarmcom