ARMmbed/mbed-os #6763: `mbed new .` fails

SeppoTakalo, 2018-04-27:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug
- Priority: Blocker

---------------------------------------------------------------
## Bug

If I create a new directory called `tmp` go into that directory and try to create new Mbed OS application, it fails:

```
tmp$ mbed new .
[mbed] Creating new program "tmp" (git)
[mbed] Adding library "mbed-os" from "https://github.com/ARMmbed/mbed-os" at branch/tag "latest"
[mbed] Updating reference "mbed-os" -> "https://github.com/ARMmbed/mbed-os/#c05d72c3c005fbb7e92c3994c32bda45218ae7fe"
[mbed] ERROR: Unknown Error: 'module' object has no attribute 'get_installed_distributions'
---
```

OS: Mac OS X

SeppoTakalo, 2018-04-27:
To fix: `pip install -U pip==9.0.3`

0xc0170, 2018-04-27:
I believe this is reported to mbed-cli and was fixed recently (pip breakage that happened recently). This one : https://github.com/ARMmbed/mbed-cli/pull/661

0xc0170, 2018-05-22:
It should be fixed via the referenced PR above.

closed 2018-05-22 by 0xc0170