ARMmbed/mbed-os #6764: LoRaWAN LoRaMacCrypto.cpp Asserts Incorrect

mattbrown015, 2018-04-27:
### Description
- Type: Bug
- Priority: Minor

'features\lorawan\lorastack\mac\LoRaMacCrypto.cpp' contains some asserts that should fire if mbedtls is incorrectly configured e.g.:

    MBED_ASSERT("[LoRaCrypto] Must enable AES, CMAC & CIPHER from mbedTLS");

I think these should actually be something like:

    MBED_ASSERT(0 && "[LoRaCrypto] Must enable AES, CMAC & CIPHER from mbedTLS");

I made a mistake with my configuration and I think I would have diagnosed the problem faster if the asserts had fired.

---------------------------------------------------------------
## Bug

**mbed-os sha:**
c8d72c524 (HEAD -> master, origin/master, origin/HEAD) Merge pull request #6693 from kjbracey-arm/equeue-ticks

**Expected behavior**

Assert with incorrect mbedtls configuration.

**Actual behavior**

Returns failure code.

**Steps to reproduce**

Disable mbedtls while using the LoRaWAN stack.

0xc0170, 2018-04-27:
@AnttiKauppila @hasnainvirk Please review

AnttiKauppila, 2018-04-27:
Arm internal ref: IOTCELL-841

mattbrown015, 2018-05-03:
If you are going to make it a compile time failure you could get rid of all the stub functions and add an #error directive.

The implication of these stubs and comments in the code is that there could be a use-case for LoRaWAN without mbedTLS but I can't think what that use-case could be.

kivaisan, 2018-05-03:
>I think MBED_STATIC_ASSERT is even better to detect the issue already during the compilation.

This is not possible after all. We must be able to build mbed-os even when lora/mbedlts is not used.

hasnainvirk, 2018-05-04:
@mattbrown015 Are you happy with the PR relating to your issue ? It will be made available in next minor release ASAP.

mattbrown015, 2018-05-08:
I'm happy if you're happy! :-)

The changes in LoRaMacCrypto.cpp look like what I was expecting and the LoRaWAN stack still works with MbedTLS included. But, I haven't investigated what happens in the various problem cases i.e. MbedTLS incorrectly configured.

mattbrown015, 2018-05-14:
Do you want me to close this?

Is it my job to close it as I opened it?

kivaisan, 2018-05-14:
@mattbrown015 yes, please close this issue.

closed 2018-05-14 by mattbrown015