ARMmbed/mbed-os #6766: HardFault while connecting with ESP8266(mbed Client example)

prvthkr, 2018-04-27:
### Description
- Type: Question
- Priority: Major

---------------------------------------------------------------
## Question

**Target**
K64F |  Custom target

**Toolchain:**
GCC_ARM

**Toolchain version:**
6.3.1 20170215 (release) [ARM/embedded-6-branch revision 245512]

**mbed-cli version:** 
1.5.1

**Expected behavior**
ESP8266 should be connect to network.

**Actual behavior**
HardFault is happing while calling easy_connect function.

**Steps to reproduce**
Same code I'm using with K64F. Everything is working properly.
like I'm able to connect with network ticker is incrementing etc.
 
But in my target it's giving hardfault. only thing I changed that is pin no. of UART pins.
My target has 192KB : RAM and 512KB : ROM.

Is this application is possible with this much of RAM(192KB) and ROM(512KB)?

And I'm  trying to connect with the [mbed connector](https://connector.mbed.com).

0xc0170, 2018-04-27:
@prvthkr We are missing the Mbed OS version, also what application version are you using?

@JanneKiiskila Any suggestions?

prvthkr, 2018-04-27:
Hello @0xc0170,

> We are missing the Mbed OS version, also what application version are you using?

Mbed OS version : [5.8.2](https://github.com/ARMmbed/mbed-os/releases/tag/mbed-os-5.8.2)

```
mbed-os-example-client (#71d23c485888)
|- easy-connect (#6b2cce9f7cef, tag: v1.2.2)
|  |- atmel-rf-driver (#ca9782e68f5f)
|  |- esp8266-driver (#c2b7e15c08ca)
|  |- mcr20a-rf-driver (#93661a696735)
|  |- stm-spirit1-rf-driver (#ce9e2f81f95f, tag: v1.0.1)
|  |- wifi-ism43362 (#d47a8c2fba0e)
|  |- wifi-x-nucleo-idw01m1 (#257d0878561b)
|  `- wizfi310-driver (#e0f7b9355e7e)
|- mbed-client (#ea04c5de7822)
|  |- mbed-client-c (#ecfa619e42b2, tag: v6.0.0)
|  |- mbed-client-classic (#4e66929607c3)
|  `- mbed-client-mbed-tls (#7e1b6d815038, tags: mbedCloudClient-R1.1, mbedCloudClient-R1.1-RC1)

```

JanneKiiskila, 2018-04-27:
You can measure the memory consumption.
1) Static RAM you can see from the compiler output.

```
Total Static RAM memory (data + bss): 78112 bytes
```
We know we're using 78 kilobytes then. 

2) Dynamic RAM consumption on top of that.
- Example for that in: https://github.com/ARMmbed/mbed-os-example-tls/pull/146

However, 192 kilobytes should be enough for that. Now, w/o having a stack back trace and knowing what "custom target" means here, it's pretty difficult to guess that the issue could be. Can you connect a debugger and get a stack backtrace?

prvthkr, 2018-04-28:
Hello @JanneKiiskila ,
 
> Static RAM you can see from the compiler output.

Total Static RAM memory (data + bss): 30408 bytes

>And [https://github.com/ARMmbed/mbed-cloud-client-example/blob/master/source/memory_tests.cpp](https://github.com/ARMmbed/mbed-cloud-client-example/blob/master/source/memory_tests.cpp) 

If I'm adding aboce file it's giving these many errors 
```
[Warning] memory_tests.cpp@126,53: format '%lu' expects argument of type 'long unsigned int', but argument 2 has type 'unsigned int' [-Wformat=]
[Warning] memory_tests.cpp@127,57: format '%lu' expects argument of type 'long unsigned int', but argument 2 has type 'unsigned int' [-Wformat=]
[Warning] memory_tests.cpp@128,73: format '%lu' expects argument of type 'long unsigned int', but argument 2 has type 'unsigned int' [-Wformat=]
[Warning] memory_tests.cpp@129,61: format '%lu' expects argument of type 'long unsigned int', but argument 2 has type 'unsigned int' [-Wformat=]
[Warning] memory_tests.cpp@130,77: format '%lu' expects argument of type 'long unsigned int', but argument 2 has type 'unsigned int' [-Wformat=]
[Warning] memory_tests.cpp@131,57: format '%lu' expects argument of type 'long unsigned int', but argument 2 has type 'unsigned int' [-Wformat=]
[Warning] memory_tests.cpp@132,61: format '%lu' expects argument of type 'long unsigned int', but argument 2 has type 'unsigned int' [-Wformat=]
[Warning] memory_tests.cpp@133,57: format '%lu' expects argument of type 'long unsigned int', but argument 2 has type 'unsigned int' [-Wformat=]
[Warning] memory_tests.cpp@134,61: format '%lu' expects argument of type 'long unsigned int', but argument 2 has type 'unsigned int' [-Wformat=]
[Warning] memory_tests.cpp@135,69: format '%lu' expects argument of type 'long unsigned int', but argument 2 has type 'unsigned int' [-Wformat=]
[Warning] memory_tests.cpp@136,71: format '%lu' expects argument of type 'long unsigned int', but argument 2 has type 'unsigned int' [-Wformat=]
[Error] memory_tests.cpp@172,5: 'delete_instance' is not a member of 'M2MSecurity'
[Warning] memory_tests.cpp@169,18: unused variable 'security' [-Wunused-variable]
[ERROR] .\memory_tests.cpp: In function 'void print_m2mobject_stats()':
.\memory_tests.cpp:126:53: warning: format '%lu' expects argument of type 'long unsigned int', but argument 2 has type 'unsigned int' [-Wformat=]
     printf("M2MBase: %" PRIu32 "\n", sizeof(M2MBase));
                                                     ^
.\memory_tests.cpp:127:57: warning: format '%lu' expects argument of type 'long unsigned int', but argument 2 has type 'unsigned int' [-Wformat=]
     printf("M2MObject: %" PRIu32 "\n", sizeof(M2MObject));
                                                         ^
.\memory_tests.cpp:128:73: warning: format '%lu' expects argument of type 'long unsigned int', but argument 2 has type 'unsigned int' [-Wformat=]
     printf("M2MObjectInstance: %" PRIu32 "\n", sizeof(M2MObjectInstance));
                                                                         ^
.\memory_tests.cpp:129:61: warning: format '%lu' expects argument of type 'long unsigned int', but argument 2 has type 'unsigned int' [-Wformat=]
     printf("M2MResource: %" PRIu32 "\n", sizeof(M2MResource));
                                                             ^
.\memory_tests.cpp:130:77: warning: format '%lu' expects argument of type 'long unsigned int', but argument 2 has type 'unsigned int' [-Wformat=]
     printf("M2MResourceInstance: %" PRIu32 "\n", sizeof(M2MResourceInstance));
                                                                             ^
.\memory_tests.cpp:131:57: warning: format '%lu' expects argument of type 'long unsigned int', but argument 2 has type 'unsigned int' [-Wformat=]
     printf("M2MDevice: %" PRIu32 "\n", sizeof(M2MDevice));
                                                         ^
.\memory_tests.cpp:132:61: warning: format '%lu' expects argument of type 'long unsigned int', but argument 2 has type 'unsigned int' [-Wformat=]
     printf("M2MFirmware: %" PRIu32 "\n", sizeof(M2MFirmware));
                                                             ^
.\memory_tests.cpp:133:57: warning: format '%lu' expects argument of type 'long unsigned int', but argument 2 has type 'unsigned int' [-Wformat=]
     printf("M2MServer: %" PRIu32 "\n", sizeof(M2MServer));
                                                         ^
.\memory_tests.cpp:134:61: warning: format '%lu' expects argument of type 'long unsigned int', but argument 2 has type 'unsigned int' [-Wformat=]
     printf("M2MSecurity: %" PRIu32 "\n", sizeof(M2MSecurity));
                                                             ^
.\memory_tests.cpp:135:69: warning: format '%lu' expects argument of type 'long unsigned int', but argument 2 has type 'unsigned int' [-Wformat=]
     printf("M2MBlockMessage: %" PRIu32 "\n", sizeof(M2MBlockMessage));
                                                                     ^
.\memory_tests.cpp:136:71: warning: format '%lu' expects argument of type 'long unsigned int', but argument 2 has type 'unsigned int' [-Wformat=]
     printf("M2MReportHandler: %" PRIu32 "\n", sizeof(M2MReportHandler));
                                                                       ^
.\memory_tests.cpp:172:5: error: 'delete_instance' is not a member of 'M2MSecurity'
     M2MSecurity::delete_instance();
     ^~~~~~~~~~~
.\memory_tests.cpp:169:18: warning: unused variable 'security' [-Wunused-variable]
[... cut ...]
This code( [SerialDeviceTerminal](https://os.mbed.com/users/mbedAustin/code/SerialDeviceTerminal/file/49fb0cbb3314/main.cpp/)) is working fine. I'm able to reset the ESP8266 module.

@SenRamakri 
> you can use the LR/PC value in the crash dump to find the point of crash. 

```
bool ESP8266::reset(void)
{
    _smutex.lock();
    for (int i = 0; i < 2; i++) {
        if (_parser.send("AT+RST")
            && _parser.recv("OK\n")
            && _parser.recv("ready")) {  // Code is hanging here. It's not able to recevie "ready"
            _smutex.unlock();
            return true;
        }
    }
    _smutex.unlock();

    return false;
}
```

SenRamakri, 2018-04-30:
Hi @prvthkr , Were you able to find exact point of crash from PC/LR values? I'm not seeing that in your above reply. Please see https://os.mbed.com/docs/v5.8/tutorials/analyzing-mbed-os-crash-dump.html for more info.

deepikabhavnani, 2018-08-24:
@prvthkr - Any updates on this issue? If issue still exists please share more details on crash as pointed by @SenRamakri

deepikabhavnani, 2018-09-18:
@prvthkr - Closing this issue for now, please reopen with more details on crash.

closed 2018-09-18 by deepikabhavnani