ARMmbed/mbed-os #6770: Support the KW40Z by back porting the KW41Z?

tesseract241, 2018-04-28:
- Type: Enhancement | Question

The KW40Z is a fundamental part of the HEXIWEAR module, providing it with bluetooth functionality and touch buttons, its only physical input, but is only supported right now through a binary built in IAR Embedded Workbench for ARM that provides pre-determined functionalities.
The KW41Z is a (partly) supported component by mbed-os which shares many hardware features with the KW40Z.

Is it possible to make use of the KW41Z target as a starting point to add a KW40Z target?

If it's easy enough and there's a guide for it, I might even attempt it myself, but I've never touched mbed or rtos internal code.

0xc0170, 2018-05-04:
@ARMmbed/team-nxp Please review

tesseract241, 2018-08-17:
@maclobdell 
My question was about the KW40Z and if the code for the KW41Z could be ported to it.  
Although what you said is very welcome news, it doesn't answer it.

maclobdell, 2018-08-17:
hi - I think it is still unknown if KW41Z code could be easily ported to KW40Z.  Perhaps that would be a good question on the NXP forums.  Sorry would couldn't answer that here.

maclobdell, 2018-08-28:
Closing in favor of moving the discussion to nxp forums.  Please reopen if you feel this is a valid issue for Mbed OS.

closed 2018-08-28 by maclobdell