ARMmbed/mbed-os #6783: LoRaWAN stack - get_channel_plan( ) not working

vsupacha, 2018-05-01:
I am working on the addition of LC3-LC8 of AS923 band. The stack only enables LC1 and LC2 which are mandatory. 

## Bug
Call LoRaWANInterface.get_channel_plan( ) always return LORAWAN_STATUS_NOT_INITIALIZED even the node can connect and send to gateway. 

**Target**
b-L072Z-LRWAN1

**Toolchain:**
mbed program exported to Keil UVision

**Toolchain version:**
MDK-ARM 5.24.1 
mbed OS 5.8.3

**Expected behavior**
Return LORAWAN_STATUS_OK with data filled in passed struct variable.

**Actual behavior**
Return LORAWAN_STATUS_NOT_INITIALIZED 

**Steps to reproduce**
Put the following code inside case CONNECTED and TX_DONE in lora_event_handler( )
retcode = lorawan.get_channel_plan(channel_plan);
if (retcode == LORAWAN_STATUS_OK ) {
  printf("Channel: %d\n", channel_plan.nb_channels);
  for (int i = 0; i < channel_plan.nb_channels; i++) {
    printf("ID: %d\n", channel_plan.channels[i].id);
  }
} else {
  printf("retcode: %d\n", retcode);
}

vsupacha, 2018-05-01:
Using debugger to investigate, I found the flow to return error code.
1. LoRaWANInterface::get_channel_plan( ) 
2. LoRaWANStack::get_enabled_channels( )
3. LoRaMac::get_channel_plan( )
4. LoRaMacChannelPlan::get_plan( )
   get_phy.attribute = PHY_CUSTOM_CHANNEL_PLAN_SUPPORT;
5. LoRaPHY::get_phy_params( )
   case PHY_CUSTOM_CHANNEL_PLAN_SUPPORT:
       phyParam.value = (uint32_t) phy_params.custom_channelplans_supported;

Searching source code, I found that only the value is initialized as <true> for a limited set of PHY, i.e. PHYKR920, PHYIN865, PHYEU868, PHYEU433, PHYCN779. Unless the value is true, the modification of frequency sub-bands is not possible,

hasnainvirk, 2018-05-02:
The bug is real. We will patch it shortly.

vsupacha, 2018-05-02:
I have tested set_channel_plan( ) function with the following code.

`lorawan_channelplan_t channel_plan;`
`loramac_channel_t channels[16];`
`channel_plan.nb_channels = 6;`
`channel_plan.channels = channels;`
`channel_plan.channels[0].id = 2;`
`channel_plan.channels[0].ch_param = AS923_LC3;`
`lorawan.set_channel_plan(channel_plan);`

At first, it seem to work because I can use get_channel_plan( ) to verify new channels were added. But the connectivity became unreliable. It can only send 1 or 2 messages and then timeout. I have tested along with similar board running ST I-Cube example code without any problem. 

If I can confirm this, I will add another issue later.

hasnainvirk, 2018-05-04:
@vsupacha If you are happy with the contents of the PR relating to your issue, please close this issue. Otherwise, let us know what else you have in mind.

hasnainvirk, 2018-05-07:
@vsupacha I am struggling to understand what you mean LC3-LC8 ? I checked the link you sent, it does not say anywhere that there are more default channels other than LC1 and LC2. The spec clearly says, you have two default channels, LC1 and LC2. 7 out of 8 other normal data channels, are LoRa modulated and the last channel is FSK modulated.

vsupacha, 2018-05-07:
@hasnainvirk  I refer to the list of uplink frequencies of AS923 starting from 923.2MHz and defined in the LoRaPHYAS923.cpp. 

`static const channel_params_t AS923_LC1 = { 923200000, 0, { ( ( DR_5 << 4 ) | DR_0 ) }, 0 };`
`static const channel_params_t AS923_LC2 = { 923400000, 0, { ( ( DR_5 << 4 ) | DR_0 ) }, 0 };`

I used LC3-LC8 for the uplink frequencies from 923.6MHz to 924.6MHz as described by
https://www.thethingsnetwork.org/docs/lorawan/frequency-plans.html#as923-925

hasnainvirk, 2018-05-08:
@vsupacha  LC1 and LC2 means default channels which are used to send Join Requests. Once Joining is complete, these channels can be used for data traffic as well. However, you cannot send Join Requests on any other channels than LC1 and LC2. The remaining channels can either be set by an optional CFlist that comes from the network or you can set them in your custom channel plan. Please note that if your network sends a CFlist, it will replace any custom channel plan you have defined.  So in that sense, after the fix you proposed, you should have a working and standard compliant solution. Hope it has helped you. Please let us know if you have anymore questions.

vsupacha, 2018-05-08:
Yes, I understand that point. That is why I would like to do add_channel_plan() in the event CONNECTED. At present, I am working with both mbed LoRaWAN stack and ST-ICUBE stack. The difference are as follows.
1. the method to handle the period for duty cycle is different. ST-ICUBE stack allows me to configure the period in the code, while mbed stack hides this setting somewhere.
2. when node is connected to the network. Node with ST-ICUBE stack seem to jump around LC3-LC8 as shown by the logging screen at network server side. After add_channel_plan() for LC3-LC8 and listed by get_channel_plan(), mbed stack seem to stay on LC1-LC2 with LC3 occasionally.
3. After the modification, nodes with mbed stack became unreliable to connect with network server. 

I plan to investigate more after a series of urgent jobs in this week.

vsupacha, 2018-06-03:
I have confirmed using a RisingHF gateway connected to the loriot.io server using OTAA method that mbed 5.8.5 LoRaWAN stack can extend to other frequency bands. But I found another problem with the output from get_channel_plan(). So I have opened another issue to report what happened.