ARMmbed/mbed-os #6789: Adding a core dump utility

netanelgonen, 2018-05-02:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Enhancement 
- Priority: Major 

## Enhancement

Adding a core dump utility will help to debug and will make developing much faster.
using a core dump will let bugs contain the core dump file and reproducing will be easier.

netanelgonen, 2018-05-02:
Hi @0xc0170  a full core dump contain the location of the fail a large stack trace and the memory.
a core dump you can load a debugger and see the full system data while the crash happened just like debugging and stepping to the place the crash happened

SenRamakri, 2018-05-02:
Hi @netanelgonen - Are you requesting this enhancement or do you have a design/implementation for this? And where/how do you intent to generate the core dump?

netanelgonen, 2018-05-02:
Hi @SenRamakri , This is a request...

netanelgonen, 2018-05-02:
@SenRamakri better?

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCORE-342*

closed 2019-10-31 by ciarmcom