ARMmbed/mbed-os #6790: EasyCellularConnection fails to connect without debugging

vidavidorra, 2018-05-02:
### Description
- Type: Bug
- Priority: Blocker

---------------------------------------------------------------
## Bug

**Target**
Custom target for STM32F205RG (see [here](https://github.com/withthegrid/mbed-os/tree/wtg-mbed-os/targets/TARGET_STM/TARGET_STM32F2/TARGET_STM32F205RG)

**Toolchain:**
GCC_ARM

**Toolchain version:**
7.2.1 20170904

**mbed-cli version:** 
1.5.0

**mbed-os sha:**
Fork from mbed-os (192a8e5094bf5eb2ecebc8c0dc5470bb157c790f)
Actual used sha from the fork ([9f3347e](https://github.com/withthegrid/mbed-os/tree/9f3347ee258e67eee88fa3f1066098e1dc8b974e))

**Expected behavior**
Like EasyCellularConnection, connection in around 20 seconds or giving an error return after around 60 seconds if I recall the behaviour correctly.

**Actual behavior**
Using the `EasyCellularConnection`, because the `OnboardCellularInterface` is going to be deprecated in 5.9.0. When using the OnboardCellularInterface, making the connection generally takes around 20 seconds but with the EasyCellularConnection this takes somewhere between 40 and 120 seconds or does not connect at all without any apparent reason. When it does not connect at all, it blocks the program without returning anything.
The strangest thing is that when I initialise EasyCellularConnection with debugging on (passing `true` as argument to the constructor) the connection is repeatedly made in around 20 seconds, the same behaviour as when I use EasyCellularConnection.
Since this only happens when I don't use the debugging, I cannot really add any logging here about the behaviour when it doesn't connect and blocks.

I've added the debug log of EasyCellularConnection with debugging on in case that might be any help. The `onboard_modem...` and `CellularModem` are debugging I've added in the low-level cellular implementation of our target and shows the same behaviour as when I use OnboardCellularInterface. 

```
[Connection] Creating cellular interface...
EasyCellularConnection::connect()
 . onboard_modem_init()
    CellularModemInit()
    CellularModemPowerEnable()
 . onboard_modem_power_up()
    CellularModemPulsePowerOn()

----------readable----------: 1
¸
----------readable----------

----------readable----------: 1
A
----------readable----------

----------resp buff:----------
A
----------buff----------

----------readable----------: 5
ATE0A
----------readable----------

----------readable----------: 1
A
----------readable----------

----------resp buff:----------
A
----------buff----------

----------readable----------: 2
AT
----------readable----------

----------readable----------: 3
ATE
----------readable----------

----------readable----------: 5
ATE0
----------readable----------

----------readable----------: 14
ATE0

ERROR

----------readable----------

----------readable----------: 1

----------readable----------

----------resp buff:----------

----------buff----------

----------readable----------: 6

OK

----------readable----------

----------readable----------: 1

----------readable----------

----------resp buff:----------

----------buff----------

----------readable----------: 6

OK

----------readable----------

----------readable----------: 1

----------readable----------

----------resp buff:----------

----------buff----------

----------readable----------: 16

u-blox

OK

----------readable----------

----------read_string buff:----------
u-blox

OK

----------buff----------

----------readable----------: 1

----------readable----------

----------resp buff:----------

----------buff----------

----------readable----------: 13

SARA-G350

----------readable----------

----------read_string buff:----------
SARA-G350

----------buff----------

----------readable----------: 6

OK

----------readable----------

----------readable----------: 1

----------readable----------

----------resp buff:----------

----------buff----------

----------readable----------: 15

08.90

OK

----------readable----------

----------read_string buff:----------
08.90

OK

----------buff----------

----------readable----------: 1

----------readable----------

----------resp buff:----------

----------buff----------

----------readable----------: 16

+CPIN: READY

----------readable----------

----------read_string buff:----------
READY

----------buff----------

----------readable----------: 6

OK

----------readable----------

----------readable----------: 1

----------readable----------

----------resp buff:----------

----------buff----------

----------readable----------: 6

OK

----------readable----------

----------readable----------: 1

----------readable----------

----------resp buff:----------

----------buff----------

----------readable----------: 16

+CGREG: 2,5,"0
----------readable----------

----------read_string buff:----------
2,5,"0
----------buff----------

----------read_string buff:----------
5,"0
----------buff----------

----------read_string buff:----------
"0
----------buff----------

----------readable----------: 16
0DE","F544"

O
----------readable----------

----------read_string buff:----------
"F544"

O
----------buff----------

----------readable----------: 3
K

----------readable----------

----------readable----------: 1

----------readable----------

----------resp buff:----------

----------buff----------

----------readable----------: 6

OK

----------readable----------

----------readable----------: 1

----------readable----------

----------resp buff:----------

----------buff----------

----------readable----------: 16

+CGATT: 1

O
----------readable----------

----------read_string buff:----------
1

O
----------buff----------

----------readable----------: 3
K

----------readable----------

----------readable----------: 1

----------readable----------

----------resp buff:----------

----------buff----------

----------readable----------: 6

OK

----------readable----------

----------readable----------: 1

----------readable----------

----------resp buff:----------

----------buff----------

----------readable----------: 6

OK

----------readable----------

----------readable----------: 1

----------readable----------

----------resp buff:----------

----------buff----------

----------readable----------: 11

CONNECT

----------readable----------
    stack ( start: 20009848 end: 2000C848 size: 00003000 used: 00000890 ) thread ( id: 2000C89C entry: 08063819 name: main_thread )
    stack ( start: 20004280 end: 20004480 size: 00000200 used: 00000070 ) thread ( id: 200041EC entry: 0806395D name: unknown )
    stack ( start: 2000F210 end: 20010210 size: 00001000 used: 00000378 ) thread ( id: 2000140C entry: 080187D9 name: tcpip_thread )
    stack ( start: 20004480 end: 20004780 size: 00000300 used: 00000098 ) thread ( id: 20004234 entry: 08068805 name: unknown )
    stack ( start: 200103B0 end: 200109B0 size: 00000600 used: 00000278 ) thread ( id: 20010360 entry: 08068D3B name: application_unnamed_thread )
    stack ( start: 2000E1E0 end: 2000F1E0 size: 00001000 used: 00000C80 ) thread ( id: 2000E190 entry: 08068D3B name: application_unnamed_thread )
     heap ( start: 2000CB5C end: 2001FC00 size: 000130A4 used: 00000000 )  alloc ( ok: 00000000  fail: 00000000 )
isr_stack ( start: 2001FC00 end: 20020000 size: 00000400 used: 00000400 )
[Connection] Connected to Network successfully
[Connection] IP address 10.9.115.62
```

**Steps to reproduce**
I've added the code here. Note that I use the `OnboardCellularInterface` here, which is a typedef to `EasyCellularConnection` because I have `CELLULAR_DEVICE` defined as `UBLOX_PPP` in `features/cellular/framework/common/CellularTargets.h`

```c++
#include "mbed.h"
#include "OnboardCellularInterface.h"

int main() {
    OnboardCellularInterface cellular(false);
//    EasyCellularConnection cellular;

    PRINTF("[Connection] Creating cellular interface...\n");
    cellular.set_sim_pin(MBED_CONF_APP_CELLULAR_SIM_PIN);
    cellular.set_credentials(MBED_CONF_APP_CELLULAR_APN, MBED_CONF_APP_CELLULAR_USERNAME, MBED_CONF_APP_CELLULAR_PASSWORD);

    int connect_success = cellular.connect();
    NetworkInterface* network_interface = &cellular;

    if(connect_success == 0) {
        printf("[Connection] Connected to Network successfully\n");

        const char *ip_addr  = network_interface->get_ip_address();
        if (ip_addr == NULL) {
            printf("[Connection] ERROR - No IP address\n");
        } else {
            printf("[Connection] IP address %s\n", ip_addr);
        }
    } else {
        printf("[Connection] Connection to Network Failed %d!\n", connect_success);
    }
```

AnttiKauppila, 2018-05-02:
ARM internal ref: https://jira.arm.com/browse/IOTCELL-842

vidavidorra, 2018-05-02:
@AriParkkila I'll update our fork to the latest mbed-os release (5.8.3), or is it better to go to the latest master? I see there are over 700 commits to master since mbed-os-5.8.3.
I'll post the results of this here once I got them

jarvte, 2018-05-21:
@vidavidorra any updates? I could not reproduce this one. Have you tried updating mbed-os master or enabling trace?

vidavidorra, 2018-05-22:
@jarvte I'm sorry, have been too busy with other high priority work. I'll try to test the two suggestions by @AriParkkila this week and report back the results.

vidavidorra, 2018-06-01:
Just tested with latest master (8c26e46) and same behaviour. Will give it another try with trace enabled though.

jarvte, 2018-08-14:
@vidavidorra I'm closing this issue as last comments are 2 months ago. Please open again if still an issue.

AriParkkila, 2018-08-28:
@0xc0170, @vidavidorra please close this issue. Can be reopened if the problem occurs again.

closed 2018-08-28 by vidavidorra