ARMmbed/mbed-os #6801: LoRaWAN Transceiver Activity Indication

mattbrown015, 2018-05-03:
### Description
- Type: Enhancement
- Priority: Minor

----------------------------------------------------------------
## Enhancement

Add a LoRaWAN event to indicate that a packet is about to be transmitted.

Calling `LoRaWANInterface::send` schedules a message for transmission but gives no indication of when the transmission will occur.  At DR0 in Europe the transmission may be scheduled for several minutes in the future.

We would like to flash an LED while the transceiver is active and turning on the LED at the same time as calling `LoRaWANInterface::send` doesn't feel like the correct thing to do.  If there was a `TX_START` event we could turn on the LED and then turn it off at `TX_DONE`.

At present `LoRaWANInterface` is nice and simple but I think there is a general issue that it quickly becomes inadequate.  In other words, I'm frustrated that I can't find out things like the current DR, off air time and if a sub-band is available now.

0xc0170, 2018-05-03:
@AnttiKauppila @hasnainvirk Please review

mattbrown015, 2018-05-03:
I completely agree, I like the simple API, but at the same time feel like knowing some of the metadata would help our app use LoRaWAN efficiently.  Contrary, I know! :-)

The other use I had in mind for the mythical `TX_START` was disabling the RF interface of our NFC tag but you've made me question that now.

I'll discuss it some more with the other members of the team here and see how we get on.

Thanks!

hasnainvirk, 2018-05-03:
@mattbrown015 yeah, its always good to have some meta data but we can't be sure where  is the limit to it. Someone will always need more :) 
We will surely discuss this internally, and think about what to do about meta data. Unfortunately we are handicapped in terms of memory, so we can't post too many events to the application, but there may be a way to get non-sensitive metadata via an API.

mattbrown015, 2018-05-08:
I think we've had a good discussion so I'll close the issue and also provide another illustration of the the thoughts I'm wrestling with...

We're considering a logging device that has to report measurements with a configurable sample interval.

When the sample interval is several hours and RF conditions are good the code is simple; join, start sampling, send again if sample not acknowledge, etc.

When the sample interval is minutes and RF conditions are poor things get more complicated. I'm trying to create a plan whereby we can start the join, start sampling, rejoin if necessary, consolidate samples, drop samples if we're getting behind, flash the LED, etc.

Will LoRaWAN meta-data help? I'm not sure, but if it had been available I could have investigated it.

Thanks,
Matt

closed 2018-05-08 by mattbrown015