ARMmbed/mbed-os #6804: [NRF52] implicit functions delcarations `app_util_disable_irq` cluttering compilation

drahnr, 2018-05-03:
### Description
- Type: Bug
- Priority: Minor

---------------------------------------------------------------
## Bug

**Target**
NRF52

```
[Warning] nordic_critical.c@38,5: implicit declaration of function 'app_util_disable_irq'; did you mean '__disable_irq'? [-Wimplicit-function-declaration]
[Warning] nordic_critical.c@52,5: implicit declaration of function 'app_util_enable_irq'; did you mean '__enable_irq'? [-Wimplicit-function-declaration]
[Error] nordic_critical.c@59,13: 'state_saved' undeclared (first use in this function)
[Warning] nordic_critical.c@60,1: control reaches end of non-void function [-Wreturn-type]
[ERROR] ./mbed-os/targets/TARGET_NORDIC/TARGET_NRF5/nordic_critical.c: In function 'hal_critical_section_enter':
./mbed-os/targets/TARGET_NORDIC/TARGET_NRF5/nordic_critical.c:38:5: warning: implicit declaration of function 'app_util_disable_irq'; did you mean '__disable_irq'? [-Wimplicit-function-declaration]
     app_util_disable_irq();
     ^~~~~~~~~~~~~~~~~~~~
     __disable_irq
./mbed-os/targets/TARGET_NORDIC/TARGET_NRF5/nordic_critical.c: In function 'hal_critical_section_exit':
./mbed-os/targets/TARGET_NORDIC/TARGET_NRF5/nordic_critical.c:52:5: warning: implicit declaration of function 'app_util_enable_irq'; did you mean '__enable_irq'? [-Wimplicit-function-declaration]
     app_util_enable_irq();
     ^~~~~~~~~~~~~~~~~~~
     __enable_irq
./mbed-os/targets/TARGET_NORDIC/TARGET_NRF5/nordic_critical.c: In function 'hal_in_critical_section':
./mbed-os/targets/TARGET_NORDIC/TARGET_NRF5/nordic_critical.c:59:13: error: 'state_saved' undeclared (first use in this function)
     return (state_saved != 0);
             ^~~~~~~~~~~
./mbed-os/targets/TARGET_NORDIC/TARGET_NRF5/nordic_critical.c:59:13: note: each undeclared identifier is reported only once for each function it appears in
./mbed-os/targets/TARGET_NORDIC/TARGET_NRF5/nordic_critical.c:60:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
```

drahnr, 2018-05-03:
[extern_app_util_star_irq.diff.txt](https://github.com/ARMmbed/mbed-os/files/1971146/extern_app_util_star_irq.diff.txt)

drahnr, 2018-05-04:
Related #6802

marcuschangarm, 2018-05-04:
The critical section implementation has been rewritten for Mbed OS 5.9.

cmonr, 2018-08-02:
@drahnr Could you confirm that this has been resolved?

drahnr, 2018-08-02:
yes, resolved with 5.9, just happened to migrate this week.

closed 2018-08-02 by drahnr