ARMmbed/mbed-os #6811: mbed-block Equivalent?

AGlass0fMilk, 2018-05-03:
### Description
- Type: Question
---------------------------------------------------------------
## Question

This concerns the now 3-year-old mbed-block module (https://github.com/ARMmbed/mbed-block). I like the idea of shared pointers and dynamically allocated memory blocks that deallocate themselves when lost. This is what I assume the library takes care of. I'm currently reading through the ble-ancs-client example (https://github.com/ARMmbed/ble-ancs-client) and it makes use of this library. It seams really useful for creating portable BLE services.

Is there any equivalent currently being used in mbed-os officially?

@marcuschangarm

0xc0170, 2018-05-04:
@davidsaada You might be also interested in this one

marcuschangarm, 2018-05-04:
Unfortunately, we have neither shared pointers nor the Block library in Mbed OS officially. 

It wouldn't take much effort to make it work on Mbed OS 5 though, if you would like to submit a PR to the other repo. For reference, the SharedPointer class is here: https://github.com/ARMmbed/core-util/blob/master/core-util/SharedPointer.h

closed 2018-07-06 by AGlass0fMilk