ARMmbed/mbed-os #6812: nRF5x BLE SecurityManager Implementation

AGlass0fMilk, 2018-05-04:
@marcuschangarm 

Target: **MCU_NRF52832**
**I'm using Softdevice S132**

I'm trying to port the ANCS client example that was made a few years ago with an older version of mbed. This example appears to be made when the BLE API wasn't filled out all the way. Link here:

https://github.com/ARMmbed/ble-ancs-client

I have the application running, but the device won't pair with the iPhone I've been testing with. It appears that encryption is never being established so the app can't get to actually subscribing to the ANCS. The return value of ble.securityManager().init() is "BLE_ERROR_NOT_IMPLEMENTED".

In my debugger it appears to be calling the base class SecurityManager::init() method (and thus returns that is hasn't been implemented). Not sure why it's not calling the subclass init method?

https://github.com/ARMmbed/mbed-os/blob/e43d21d4ef6025e94e4567961cb87ed2f0474a26/features/FEATURE_BLE/targets/TARGET_NORDIC/TARGET_NRF5x/source/btle/btle_security.cpp#L16-L21

pan-, 2018-05-04:
This is an error on our end; new implementation of the security manager for Nordic has been pushed back at the last minute for 5.8 (it will be present for 5.9). Unfortunately the `init` function has a new parameter now and it hasn't been propagated to the old implementation.

marcuschangarm, 2018-05-04:
> I'm trying to port the ANCS client example that was made a few years ago with an older version of mbed. 

That's awesome! 😄

AGlass0fMilk, 2018-07-06:
Should be resolved now that 5.9 is released.

closed 2018-07-06 by AGlass0fMilk