ARMmbed/mbed-os #6813: Support configuring stack size in RTOS-less build

ccli8, 2018-05-04:
### Description
- Type: Enhancement
- Priority: Major

**mbed-os sha:**
mbed-os-5.8.3

I am evaluating Mbed Cloud 1.3.1 on **NUMAKER_PFM_M487**/**GCC_ARM**. In [mbed-bootloader](https://github.com/ARMmbed/mbed-bootloader), it is built with RTOS disabled. With stack size hard-coded to 0x800 in linker script file, I meet stack overflow on `printf()` call. For RTOS enabled, I can change main thread's stack size through `mbed_app.json` (`MBED_CONF_APP_MAIN_STACK_SIZE`). But for RTOS disabled, it seems there's no way except modifying linker script file.

0xc0170, 2018-05-04:
@SenRamakri @bulislaw Please review

bulislaw, 2018-05-08:
Hi @ccli8 you are right it doesn't make sense to have 2k stack and huge heap. We should be able to configure it with some sane default. We'll try to have a look into that soonish.

andrewleech, 2018-05-10:
On an barely related note, is there an official easy to build without RTOS? 
Is `.mbedignore` the rtos folder it, or is there a better way?

bulislaw, 2018-05-11:
Right now we don't support it officially. We acknowledge it's a legitimate use case so we try not to break it, but we don't actively maintain it. We may revisit that for next major release.

0xc0170, 2018-07-13:
@ccli8 Isn't this trying to address this issue : #7238 ?

ccli8, 2018-07-16:
@0xc0170 Yes, #7238 is addressing the issue.

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCORE-261*

ccli8, 2018-12-12:
Done with #8039

closed 2018-12-12 by ccli8