ARMmbed/mbed-os #6816: HardFault_Handler BusFault_Handler and others are not weak

drahnr, 2018-05-04:
### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
NRF52_DK

**Toolchain:**
GCC_ARM

**mbed-cli version:** 
1.5.1

**mbed-os sha:**
5.8.3

**Expected behavior**
Custom defined handlers shall take precedence over the default ones

**Actual behavior**
Compiler error due to redfined symbolds

**Steps to reproduce**
Define a custom HardFault_Handler, such as i.e.

```
__attribute__ ((naked)) void HardFault_Handler (void) {
    __asm volatile(
"..."
)
}
```
and observe:
```
/home/xyz/project/./src/DeathAndDecay.cpp:246: multiple definition of `HardFault_Handler'
BUILD/XYZ/GCC_ARM/mbed-os/rtos/TARGET_CORTEX/TARGET_CORTEX_M/TOOLCHAIN_GCC/except.o:/home/xyz/project/except.S:50: first defined here
collect2: error: ld returned 1 exit status
[ERROR] BUILD/XYZ/GCC_ARM/src/DeathAndDecay.o: In function `HardFault_Handler':
/home/xyz/project/./src/DeathAndDecay.cpp:246: multiple definition of `HardFault_Handler'
BUILD/XYZ/GCC_ARM/mbed-os/rtos/TARGET_CORTEX/TARGET_CORTEX_M/TOOLCHAIN_GCC/except.o:/home/xyz/project/except.S:50: first defined here
collect2: error: ld returned 1 exit status
```

Attempting to use
`-D MBED_FAULT_HANDLER_DISABLED` does not help.

drahnr, 2018-05-04:
It should suffices to change the line
```
.global HardFault_Handler
```
to
```
.weak HardFault_Handler
```
(for GCC!)
in `except.S`

0xc0170, 2018-05-04:
@SenRamakri Please review

drahnr, 2018-05-09:
I can not reproduce this right now and code should be fine with just `-DMBED_FAULT_HANDLER_DISABLED=1`, so I am closing this.

closed 2018-05-09 by drahnr