ARMmbed/mbed-os #6820: BLE addressType in AdvertisementCallbackParams_t, ConnectionCallbackParams_t and DisconnectionCallbackParams_t is inconsistent

RobMeades, 2018-05-04:
### Description
- Type: Question
- Priority: Minor

I have some BLE code in which my `Gap::CENTRAL` device looks for nearby devices and connects to them to read their characteristics.  In my advertisement callback, when I call `connect()` on a device, I use the `peerAddr` and `addressType` that I was given in `AdvertisementCallbackParams_t` (which always seems to be `RANDOM_STATIC (1)`).  However, for _some_ (but not all) BLE peer devices, when my connection callback is called, the `peerAddrType` given to me in `ConnectionCallbackParams_t` changes: it becomes `RANDOM_PRIVATE_RESOLVABLE (3)` instead.  Is this meant to happen?  An example:

```
Added BLE device 48:c2:7c:2e:d1:d1, (address type RANDOM_STATIC) to list.
BLE device 48:c2:7c:2e:d1:d1 is visible.
  Advertising payload type 0x01 (FLAGS) (1 byte(s)): 0x1a.
  ...and discoverable, attempting to connect to it (address type RANDOM_STATIC), connect() successfully issued.
BLE device 48:c2:7c:2e:d1:d1 (address type RANDOM_PRIVATE_RESOLVABLE) is now connected (handle 0).
```
I'm asking because I keep a record of the devices along with their addresses and their address types, so a change of address type means I don't find the device in my list.  I could, of course, ignore the address type (if it is going to change like this): would that be the correct thing to do?

My code is [attached](https://github.com/ARMmbed/mbed-os/files/1974492/main.cpp.txt) if it helps.

RobMeades, 2018-05-04:
Just to be clear, to `connect()` to a device (from our [other conversation](https://os.mbed.com/questions/81028/BLE-connect-hangs/?c=30089)) the address type must be `RANDOM_STATIC` so, before I call `connect()`, I should check in the `AdvertisementCallbackParams_t` that `addressType` is `RANDOM_STATIC`.  However, in terms of checking the address type of a BLE peer for matching purposes I should lump all of the `RANDOM` types together, simply checking for `RANDOM` versus `PUBLIC`. Is that correct?

RobMeades, 2018-05-04:
Understood: in that case it might be worth an update to the [GAP example](https://os.mbed.com/docs/v5.8/reference/gap.html) and [BLE_LEDBlinker](https://github.com/ARMmbed/mbed-os-example-ble/blob/master/BLE_LEDBlinker/source/main.cpp#L57) to use the `addressType` passed in as they both currently hard code the `connect()` address type.

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTPAN-232*

pan-, 2020-10-05:
I'm closing this, the address and examples have been reworked. Inclusion of proper privacy support (see https://github.com/ARMmbed/mbed-os/pull/13717) will increase consistency when dealing with private resolvable addresses.

closed 2020-10-05 by pan-