ARMmbed/mbed-os #6824: nRF5x UART Hang

AGlass0fMilk, 2018-05-04:
### Description
Hi, I'm recently synced my fork of mbed-os and pulled in the updated nRF5x serial API changes. I also noticed the base class in target.json introduced the following concerning UART HWFC:

https://github.com/ARMmbed/mbed-os/blob/6361e92d62efcc113723d43277705dbdf4594716/targets/targets.json#L15-L19

I had to change my custom target to remove the hardware flow control override (I dealt with this issue in the past, see issue ARMmbed/mbed-os-example-ble#25 ) since this is not part of the base class anymore.

I'm now seeing a hang when trying to print to a UART/serial. It's hanging at the following busy-wait loop for the TXend event:

https://github.com/ARMmbed/mbed-os/blob/6361e92d62efcc113723d43277705dbdf4594716/targets/TARGET_NORDIC/TARGET_NRF5x/serial_api.c#L1476-L1478

Is there an equivalent fix for this? My custom target does not even have hw flow control pins broken out so that's not an option.

@marcuschangarm 

(Sorry for blowing up the issues tab)

marcuschangarm, 2018-05-04:
Hey, can you post your custom_target.json, please? 
And is this with a simple Hello World test with printf and nothing else?

AGlass0fMilk, 2018-05-04:
The only relevant details from my custom target are:
`"inherits": ["HU_FAMILY_AIRBORNE", "NRF52_DK"],
	"macros_remove": ["BOARD_PCA10040"],
	"macros_add": ["ENABLE_SWO", "I2C_TIMEOUT_VALUE_US=1000", "BOARD_CUSTOM"],`

The other characteristics are for internal configuration.

The program I am running looks like this:

```
#include "mbed_debug.h"
#include "mbed_wait_api.h"

int main(void)
{
	/** Initialization and thread spinoff goes here */

	// Latch the device power on
	power_enable_pin = 1;

	while(true)
	{
		/** Application code goes here */
	       debug("hello!");
	       wait_ms(1000);
	}
}
```

AGlass0fMilk, 2018-05-04:
I'm testing the same code on a fresh mbed-os program for the actual NRF52_DK board and I'm getting similar results... no serial output.

AGlass0fMilk, 2018-05-04:
Update: I added the following to my mbed_app.json and it fixed the hang:
`"target.uart_hwfc": 0`

mbed-cli was complaining before that I was trying to override a config parameter that didn't exist (since this was removed from the official mbed targets.json file) but it seems that the code still uses it somewhere?

AGlass0fMilk, 2018-05-04:
For some reason that doesn't work for my custom_target... the stdio_uart still has HWFC enabled... is there a way to disable hardware flow control???

marcuschangarm, 2018-05-04:
Ah, I see. You probably want to inherit from `MCU_NRF52832` directly and not from `NRF52_DK`. Flow control is disabled by default but the `NRF52_DK` enables flow control for itself.

marcuschangarm, 2018-05-04:
Depending on your chip revision you should add (some of) these macros as well:

https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_NORDIC/TARGET_NRF5x/mbed_lib.json#L87-L107

AGlass0fMilk, 2018-05-04:
Thank you for the fix @marcuschangarm... I changed over to inheriting from MCU_NRF52832 and added the errata fix macros.

closed 2018-05-04 by AGlass0fMilk