ARMmbed/mbed-os #6827: STM32F4 - I2C slave bug

YarivCol, 2018-05-06:
### Description
Type: Bug
- Related issue: `#abc`
- Priority: Minor

---------------------------------------------------------------
## Bug

**Target**
stm32f4

**Toolchain:**
IAR

**Toolchain version:**

**mbed-cli version:** 
(`mbed --version`)

**mbed-os sha:**
(`git log -n1 --oneline`)

I am using my stm32f4 microcontroller as an i2c slave for the first time.
At the begining, I used the example of i2c slave from I2CSlave.h.
The I2C master did not recieve the message sent by the i2c slave, so I used my scope to see the clk and data.
I noticed that the slave (my microcontroller stm32f4) is pulling down the clk when he recieves a read massage.
Looking at the datasheet and the registers made I saw that the NOSTRETCH register is disabled as defualt in i2c_frequency() at i2c_api.c
Changing the register in the fuction to enable solved my problem.
I would like to know why isn't there an api for the register and why is it disabled as default?

0xc0170, 2018-05-07:
@ARMmbed/team-st-mcd Please review

bcostm, 2018-05-07:
> I would like to know why isn't there an api for the register

What do you mean by "register" ? An API to access any MCU register ? or in your case an API to control the I2C stretch functionality ? If this is the last case, then there is no possibility in the mbed API to control it.

> why is it disabled as default?

As there is no possibility in the mbed API to control it, we had to make a choice. But it is very simple to change it and enable the strech per default. But I don't know which impact it could have on all I2C devices connected ?

bcostm, 2018-06-12:
After internal discussion, we think the best thing would be to have an API in mbed to change this stretch mode. For example in i2c_init or create a new method set_stretchmode ?

screamerbg, 2018-09-10:
@bulislaw @donatieng @ithinuel See answer from @bcostm above. Also @0xc0170 - is this a bug or feature request?

ithinuel, 2018-09-11:
The strech mode is basically a flow control technique that is sensible to me to have enabled by default.
Introducing a way to disable such feature could be part of the work @scartmell-arm is doing on I²C hal API : #7834

LMESTM, 2018-09-17:
@screamerbg 
As recommended by @ithinuel  the stretch mode is indeed enabled by default on STM32 platforms s- to there is no bug in STM32.  
As @YarivCol  needs to disable it (probably the master device in the use case does not support stretch mode), then the point needs to be addressed first @ MBED API level to provide the way to disable it.
Can you update the ticket labels accordingly (e.g; removing device:st )

bcostm, 2018-09-17:
I would also either change the title or create a new one to specifically request an improvement in mbed API  to disable the stretch mode.

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTPART-6061*

bcostm, 2018-10-09:
Can you please remove the `devices:st` label ? Because the enhancement to change the stretch mode in mbed API is for all the targets and not only ST ones.

jeromecoutant, 2019-02-20:
> @0xc0170 removed the [devices: st]
> @elm3  added the [devices: st]

Could you correct the label and remove st device ?

0xc0170, 2019-02-20:
Removed as this is generic issue

0xc0170, 2020-10-22:
> I would also either change the title or create a new one to specifically request an improvement in mbed API to disable the stretch mode.

+1, lets get a new feature request on the forum (not here as it's not technical issue but a new feature request). I'll close this one.

closed 2020-10-22 by 0xc0170