ARMmbed/mbed-os #6835: Compilation error on NRF51822

imi415, 2018-05-08:
### Description

* Version: Mbed OS 5.8.3(release)
* Target: NRF51_DK_(BOOT|OTA)(both tried)
* Toolchain: GCC_ARM, 7(2017-q4), ARM / GCC_ARM, 8, GNU(both have exactly same error output)
* Tool: mbed-cli
---
Steps to reproduce
```
Compile [ 13.8%]: critical_section_api.c
[Error] critical_section_api.c@75,55: expected ';' before '}' token
[ERROR] ./mbed-os/features/FEATURE_BLE/targets/TARGET_NORDIC/TARGET_MCU_NRF51822/hal_patch/critical_section_api.c: In function 'hal_critical_section_exit':
./mbed-os/features/FEATURE_BLE/targets/TARGET_NORDIC/TARGET_MCU_NRF51822/hal_patch/critical_section_api.c:75:55: error: expected ';' before '}' token
         sd_nvic_critical_region_exit(_state._sd_state)
                                                       ^
                                                       ;
     } else {
     ~                                                  
```

Added a semicolon after that line did solved that issue, however it gave another error:
```
Compile [ 17.7%]: critical_section_api.c
[ERROR] /tmp/ccIT1sAp.s: Assembler messages:
/tmp/ccIT1sAp.s:34: Error: SVC is not permitted on this architecture
/tmp/ccIT1sAp.s:59: Error: SVC is not permitted on this architecture
/tmp/ccIT1sAp.s:85: Error: SVC is not permitted on this architecture
```
Added `-march=armv6s-m` to compiler options (by using custom `--profile`)  solved this, but I had no idea how this worked, _OTA succeeded while _BOOT still failing at link stage.
```
[ERROR] Starting addresses are different
```
### Issue request type

[ ] Question  
[ ] Enhancement  
[x] Bug

0xc0170, 2018-05-08:
@marcuschangarm Please review

marcuschangarm, 2018-05-08:
@scartmell-arm @bulislaw 

Looks like something isn't right here: https://github.com/ARMmbed/mbed-os/blame/master/features/FEATURE_BLE/targets/TARGET_NORDIC/TARGET_MCU_NRF51822/hal_patch/critical_section_api.c#L75

bulislaw, 2018-05-09:
@scartmell-arm please have a look at this

closed 2018-05-09 by cmonr

theotherjimmy, 2018-05-09:
ahahahah I'll close my PR

PR #6849 (2018-05-09): Add missing semicolon to NRF51 critical_section implementation
### Description

Add missing semicolon to FEATURE_BLE critical section implementation.

Fixes: #6835

### Pull request type

    [X] Fix
    [ ] Refactor
    [ ] New target
    [ ] Feature
    [ ] Breaking change