ARMmbed/mbed-os #6840: STM32L0 EventQueue LPTIMER

simonnilsson, 2018-05-08:
### Description

Target: Custom stm32l071 target based on the existing stm32l072 target
Tools: GCC_ARM, Keil uVision
Mbed OS Version: ARMmbed/mbed-os@c8d72c5

Using the following code:

```

### Issue request type
<!-- 
    Required
    Please add only one X to one of the following types. Do not fill multiple types. (Split the issue otherwise.)
    Please note this is not a GitHub task list; indenting the boxes or changing the format to add a '.' or '*' in front
    of them changes the meaning incorrectly. The only changes to make are to add a description under the
    description heading and to add an 'x' to the correct box.

    [X] Question
    [ ] Enhancement 
    [ ] Bug 
-->

[ ] Question  
[ ] Enhancement  
[X] Bug

0xc0170, 2018-05-09:
@simonnilsson The issue is where? Is it hal implementation or generic layer (ticker) ? From lorawan reference, this was reproduced on the another hw, seems like ticker issue

@kivaisan Can you confirm that this also affects another platforms?

simonnilsson, 2018-05-09:
@0xc0170 I'm not sure if the problem is in the target, generic layer or even me doing something wrong. 

I believe @kivaisan was not able to reproduce it on the platform he tried it on. It would be nice to try it on another stm32 target to see if it's specific for stm32 but I don't have one at hand unfortunatly.

kivaisan, 2018-05-11:
Yes, I tried with K64F and it worked ok but I haven't tested with any STM32L0 device yet.

kivaisan, 2018-05-11:
@simonnilsson I just tested your code with DISCO_L072CZ_LRWAN1 (STM32L072 mcu) and the event queue is still working after 1.5h. So I think the problem is in your own custom target.

simonnilsson, 2018-05-11:
Okay so I also got my hand on a **DISCO_L072CZ_LRWAN1** now. Ran the same test using the latest mbed-os as of today (https://github.com/ARMmbed/mbed-os/commit/30e39eeb10b2e0961068748651f001fe008b374a). And I'm getting the same problem on this board so we must be doing something differently.

This is my `.mbedignore`:
```
mbed-os/rtos/*
mbed-os/features/cellular/*
mbed-os/features/FEATURE_CLIENT/*
mbed-os/features/FEATURE_COMMON_PAL/*
mbed-os/features/FEATURE_UVISOR/*
mbed-os/features/frameworks/*
mbed-os/features/net/*
mbed-os/features/netsocket/*
mbed-os/features/nanostack/*
mbed-os/features/storage/*
mbed-os/features/filesystem/*
mbed-os/features/nvstore/*
mbed-os/features/cellular/*
```

And theese are the relevant entries in my `mbed_config.h`:
```
#define MBED_CONF_EVENTS_SHARED_DISPATCH_FROM_APPLICATION 0
#define MBED_CONF_EVENTS_PRESENT                          1
#define MBED_CONF_PLATFORM_FORCE_NON_COPYABLE_ERROR       0
#define MBED_CONF_TARGET_LOWPOWERTIMER_LPTIM              1
#define MBED_CONF_PLATFORM_POLL_USE_LOWPOWER_TIMER        0
#define CLOCK_SOURCE                                      USE_PLL_HSI
#define MBED_CONF_EVENTS_SHARED_HIGHPRIO_STACKSIZE        1024
#define MBED_CONF_APP_MAIN_STACK_SIZE                     3072
#define MBED_CONF_EVENTS_SHARED_STACKSIZE                 1024
#define MBED_CONF_EVENTS_SHARED_EVENTSIZE                 256 
#define MBED_CONF_DRIVERS_UART_SERIAL_TXBUF_SIZE          256
#define MBED_CONF_EVENTS_USE_LOWPOWER_TIMER_TICKER        1
#define MBED_CONF_TARGET_LSE_AVAILABLE                    1
#define MBED_CONF_EVENTS_SHARED_HIGHPRIO_EVENTSIZE        256
```

kivaisan, 2018-05-14:
@simonnilsson I was able to reproduce the problem with DISCO board with non-rtos and lowpower timers enabled (sorry for confision with my original comment which I deleted as I noticed I first forgot to enable lowpower timers).

0xc0170, 2018-05-14:
@ARMmbed/team-st-mcd Is this known and already being reported?

jeromecoutant, 2018-05-14:
Hi
This has to be checked in the ticker branch

LMESTM, 2018-05-14:
I had observed similar behavior and what I debugged at the time showed that there was some issue around managing the queue of Low power tickers. It looked to me that the Max time tickers and next programmed ticker where inverted in the list so that the max time tickers was programmed instead of the next one ... 
What we did next was to test on  ARMmbed:feature-hal-spec-ticker  and this looked like fixed.
 So could check against this branch ?

0xc0170, 2018-05-14:
@simonnilsson Can you please test that feature branch mentioned (https://github.com/ARMmbed/mbed-os/tree/feature-hal-spec-ticker)  and let us know the results ?

simonnilsson, 2018-05-14:
@0xc0170 Will do.

jeromecoutant, 2018-05-17:
> Yes it seems to work fine on that branch. Are there any plans on merging this into master soon?

@bulislaw

bulislaw, 2018-05-17:
Hopefully for 5.9 (that is it needs to land on master next week). Fingers crossed :)

simonnilsson, 2018-06-26:
Found another problem in _equeue_sema_wait_ in **equeue_mbed.cpp** if RTOS not present the ms argument is passed as a signed integer and then multiplied by 1000. This causes an overflow in some cases!

Changing this:
```
timeout.attach_us(callback(equeue_sema_timeout, s), ms*1000);
```

to this

```
timeout.attach_us(callback(equeue_sema_timeout, s), (us_timestamp_t)ms*1000);
````
Solves the problem for me, but it is still handled as a integer outside this function so might be other issues with this.

jeromecoutant, 2018-06-26:
Hi @c1728p9 @bulislaw 
Please, check comment from @simonnilsson

jeromecoutant, 2018-06-28:
@simonnilsson 

> Yes it seems to work fine on that branch

So all is in the master branch, could you close this issue ?

About the problem in equeue_sema_wait, I really think you should open a new issue for a clean analysis from ARM team

Thx

closed 2018-06-28 by 0xc0170

jeromecoutant, 2018-07-04:
Hi @simonnilsson 
Don't forget to raise an issue for the equeue_sema_wait problem :-)