ARMmbed/mbed-os #6850: Assert traces are not shown when assert expression and/or filename is long

kivaisan, 2018-05-09:
### Description

**Target:** K64F (actually the used target does not matter)
**Toolchain:** GCC_ARM
**mbed-os version:** master (11901f115b0b81f875987cb9598dfbf0d57183c7)

If the expression and/or filename of MBED_ASSERT is too long, the assert trace print does not show up or it partially shown.

This is because the maximum length for print
 `mbed_error_printf("mbed assertation failed: %s, file: %s, line %d \n", expr, file, line);` 
is only 128 characters as defined in `mbed_error_vfprintf ` as `ERROR_BUF_SIZE`.

In our case the problem is mainly with the file name as the print includes the file name with full path.

### Issue request type

[ ] Question  
[ ] Enhancement  
[X] Bug

0xc0170, 2018-05-09:
@kivaisan The error size is predefined and can't be changed? Would you propose to make it configurable? I assume the default value 128 is fine for most cases.

cc @SenRamakri

kivaisan, 2018-05-11:
@0xc0170 I don't think having such a predefined buffer size in this is case good idea because like mentioned, the problem with this one is __FILE__ macro. For example in my greentea builds, the build system seems to put a full file path to it and we can't know in what kind of paths developers keep their work areas.

Here is an example test assert (I increased the buffer to 256 bytes in order to get this print out):
`mbed assertation failed: variable1 == variable2, file: /home/kivaisan/mbedos/loramac/mbed-os-example-lorawan/mbed-os/features/lorawan/LoRaWANInterface.cpp, line 46`
Very simple assert and already 164 characters long.

So I think the quick-fix is to increase the buffer size but in order to avoid unnecessary memory/stack usage maybe this should be refactored.

cc @kjbracey-arm

kjbracey-arm, 2018-05-11:
Normally prints shouldn't be that long, but asserts are an exception.

I think the easiest quick workaround is to modify the assert to not use a single mbed_error_vfprintf. There's no reason it can't use 4 successive ones.

Then it only fails with long filenames, and not as badly, because the trailing "line 46\n" will still come out.

kjbracey-arm, 2018-05-11:
Plus you can save 2 characters by not using the "word" `assertation`.

kivaisan, 2018-05-31:
Any plans for this?

This is really problematic with greentea (I again spend some time to investigate why my test stops but actually it was again a hidden assert).

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCORE-345*

kivaisan, 2019-01-09:
Since assert now uses generic mbedos error info and expression info is shown correctly, I close this issue as fixed.

closed 2019-01-09 by kivaisan