ARMmbed/mbed-os #6853: while (1); --> SERIOUSLY? Is this error handling?

daniel-cesarini-tridonic-com, 2018-05-09:
https://github.com/ARMmbed/mbed-os/blob/68ad00ffb83b001ec371818556f5751d0052b051/features/nanostack/mbed-mesh-api/source/mesh_system.c#L58

while (1); --> SERIOUSLY? Is this error handling?

@karsev

0xc0170, 2018-05-09:
Note that we are currently working on defining error handling.

cc @SeppoTakalo please review

SeppoTakalo, 2018-05-11:
Yes, it should be `while(1);`

This is unrecoverable error and therefore the device should be stopped. S
`MBED_ASSERT();` cannot be used here, because it is compiled out in release builds.

Until better options appear, this is the correct solution.

kjbracey-arm, 2018-05-11:
There is `mbed_die()`, which we should probably use in the mbed OS port.

The `while (1)` came from a portable base implementation.

kjbracey-arm, 2018-05-11:
Mind you, the "while (1)" only effectively kills Nanostack-related components, by jamming its event loop thread. It doesn't kill the entire system, which is effectively what was originally intended by the `while (1)` for bare-metal.

So one positive suggestion for improvement. Change the code to be

     nanostack_failure();

    NS_NORETURN MBED_WEAK void nanostack_failure()
    {
         mbed_die();
    }

That kills the system by default (presumably by triggering reboot watchdog), but gives you the option to provide an alternative `nanostack_failure` implementation. You could sleep the event thread and issue an event to some other monitor thread.

Regardless it's not possible to restart Nanostack in the event of heap corruption - you will need to reboot to reset static memory.

And bear in mind that all of this is only relevant in the rare case of heap corruption being detected.

It's more likely that heap corruption causes a hard fault due to a bogus pointer, which is system death regardless.

closed 2018-08-14 by NirSonnenschein