ARMmbed/mbed-os #6854: Question: Why is the SysTick timer not used in STM32 support?

daid, 2018-05-09:
### Description
Sorry to bother here. But I noticed that the STM32 support is using hardware timers for the main RTOS timer instead of the SysTick timer. This results in certain PWM pins not being available for use.
I was wondering if there is a specific reason for not using the SysTick timer?

(Our electronics designer wasn't aware of this restriction, and now I'm stuck with a board which has a required PWM output on the timer that is used for the RTOS)

### Issue request type

[x] Question
[ ] Enhancement
[ ] Bug

0xc0170, 2018-05-09:
hardware timer for RTOS instead of Systick? Can you provide file reference?

I am aware that nordic does, as it does not have systick. STM32 implementation does not implement os tick RTOS API thus should use the default one (systick)

daid, 2018-05-09:
Maybe I mis-understand something, but this code is clearly using a hardware timer:

https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/us_ticker_32b.c#L33
https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32F3/TARGET_STM32F303x8/device/hal_tick.h#L46

0xc0170, 2018-05-09:
That is us ticker (high frequency ticker) for Timer/Ticker objects.  RTOS uses Systick. How are you getting the conflicts - what code are you using - what objects? Is it `wait()` or you using Timer/Ticker/etc ? They could cause a conflict but then it looks a limitation of the implementation?

Details about OS tick here https://www.keil.com/pack/doc/CMSIS/RTOS2/html/group__CMSIS__RTOS__TickAPI.html (look at the default implementation - that one is used for ST).

cc @ARMmbed/team-st-mcd

daid, 2018-05-09:
I'm trying to use a PWM Pin which is connected to TIM2 on the STM32F303, and we also have a few Ticker objects. So maybe it's not a conflict with the RTOS but with the Ticker/Timer implementation.

Guess that's why it's not using SysTick, to prevent a conflict with the RTOS?

0xc0170, 2018-05-09:
> Guess that's why it's not using SysTick, to prevent a conflict with the RTOS?

correct

> I'm trying to use a PWM Pin which is connected to TIM2 on the STM32F303, and we also have a few Ticker objects. So maybe it's not a conflict with the RTOS but with the Ticker/Timer implementation.

@ARMmbed/team-st-mcd Any recommendations how to use both (us ticker and pwmout) ?

LMESTM, 2018-05-09:
@daid As there are usually several HW timers available this was a design choice made long ago to use 1 dedicated HW timer for handling the timeouts in the STM32 HAL SDK layer independantly from the mbed OS ticker. 

Now fr your specific board, you may decide to use another HW timer than TIMER2 by modifying the lines. 
```
#define TIM_MST      TIM2
#define TIM_MST_IRQ  TIM2_IRQn
#define TIM_MST_RCC  __TIM2_CLK_ENABLE()
#define TIM_MST_DBGMCU_FREEZE  __HAL_DBGMCU_FREEZE_TIM2()
```
Isn't that something that could solve your case ?

daid, 2018-05-09:
I think it might, can this also work with a 16bit timer? TIMER2 seems to be the only 32bit timer on the STM32F303. There are enough free 16bit timers however, would there be a disadvantage of using a 16bit timer?

LMESTM, 2018-05-11:
Should work fine as well - you'll have to also add
```
#define TIM_MST_16BIT  1 // 1=16-bit timer, 0=32-bit timer
```
and also check
```
#define TIM_MST_PCLK  1 // Select the peripheral clock number (1 or 2)
```
Please give it a try and let us know.
 If any issue please try to reproduce on a NUCLEO board and we would investigate.

daid, 2018-05-11:
Will do. Got a bunch of Nucleos to test on in case of problems.

Would it be an idea to patch the STM32 targets to use a different timer with has no PWM conflict for the default supported mbed targets? For example, the F303 has a few timers (TIM6/TIM7) that are only useful as timer or to trigger the DAC (which isn't used in that way by the mbed API)

LMESTM, 2018-05-11:
Sounds like a good idea. This could be added to the to-do list as an enhancement request. 
cc @bcostm

daid, 2018-05-14:
@bcostm Actually, we can move around the timer in some chips without disabling any PWM. So it would give more PWMs, while keeping all the current ones, no compatibility loss. Like I said, Timer6 has no PWM outputs attached to it at all on the F303.
And I have a custom board with the F303VB, so for my custom board I can get this to work.

daid, 2018-05-15:
Just tried with TIM6 and TIM7, didn't work, those timers lack some of the features used. TIM20 works, but is not available on all F303 chips. This is quite a puzzle, I start to see why TIM2 was used.

daid, 2018-05-16:
Closing this issue, I got my answer. I think I understand why TIM2 is used, it is done to prevent trying to puzzle for each different chip which is the best timer to use.

Certain targets can be improved on this area, but it is a puzzle per chip.

closed 2018-05-16 by daid

LMESTM, 2018-05-16:
@daid - thanks for your time spent in understanding and testing - and thanks the good wrap-up !

adustm, 2018-05-17:
> thanks the good wrap-up !

Hello,
You're absolutely right. The best wrap up is : 

> This is quite a puzzle

:-)