ARMmbed/mbed-os #6858: setPairingRequestAuthorisation() doesn't prevent pairing on nRF5x

s4y, 2018-05-10:
### Description

Target: MCU_NRF52832
Toolchain: mbed 5.8

I implemented a minimal `SecurityManager::EventHandler` that looks like this:

```
class SecurityManagerEventHandler: public SecurityManager::EventHandler {
	void pairingRequest(ble::connection_handle_t connectionHandle) override {
		BLE::Instance().securityManager().cancelPairingRequest(connectionHandle);
	}
};
```

…and applied it like this:

```
	SecurityManagerEventHandler smeh;

	SecurityManager &sm = ble.securityManager();
	sm.setSecurityManagerEventHandler(&smeh);
	sm.setPairingRequestAuthorisation(true);
```

…but all pairing requests are accepted. I poked at the code, and it looks like `GenericSecurityManager::on_pairing_request()` is only called on `TARGET_CORDIO`.

Does that sound correct? If so, it would be great if this could fail the build until it's implemented. RIght now it fails insecurely by allowing all pairing.

### Issue request type

[ ] Question  
[ ] Enhancement  
[X] Bug

0xc0170, 2018-05-17:
@donatieng Please review

donatieng, 2018-06-01:
@s4y this should be fixed by f7f1272647ed5776eb77986bdad67b66e3d3f6e6. Closing this, feel free to reopen if you're still experiencing the issue.

s4y, 2018-06-04:
Thank you! In the mean time I started porting my (small) project to the Nordic SDK because I want to play with other features like OTA updates. So, I’ll close this issue, but will let you know if I run into any fallout if I try mbed again!

closed 2018-06-04 by s4y