ARMmbed/mbed-os #6861: mbed test failing to build for nrf51_dk since #6577

andrewleech, 2018-05-10:
### Description

In the last day or so `mbed test -m NRF51_DK -t GCC_ARM --verbose --clean ` has failed to build. 
None of the actual C files can compile:
```
Compile: F:\Program Files (x86)\GNU Tools ARM Embedded\6 2017-q1-update\bin\arm-none-eabi-g++ -std=gnu++98 -fno-rtti -Wvla -c -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -fmessage-length=0 -fno-exceptions -fno-builtin -ffunction-sections -fdata-sections -funsigned-char -MMD -fno-delete-null-pointer-checks -fomit-frame-pointer -Os -g1 -mcpu=cortex-m0 -mthumb -D__CORTEX_M0 -DNRF51 -D__MBED__=1 -DTARGET_LIKE_MBED -DTARGET_NRF51822 -DDEVICE_PORTINOUT=1 -D__MBED_CMSIS_RTOS_CM -DDEVICE_LOWPOWERTIMER=1 -DDEVICE_RTC=1 -DTOOLCHAIN_object -DDEVICE_SERIAL_ASYNCH=1 -D__CMSIS_RTOS -DTARGET_MCU_NRF51822_UNIFIED -DTOOLCHAIN_GCC -DDEVICE_SPISLAVE=1 -DTARGET_CORTEX_M -DTARGET_SDK11 -DDEVICE_I2C_ASYNCH=1 -DARM_MATH_CM0 -DTARGET_UVISOR_UNSUPPORTED -DTARGET_NRF5 -DFEATURE_BLE=1 -DTARGET_M0 -DTARGET_MCU_NRF51 -DCMSIS_VECTAB_VIRTUAL -DTARGET_MCU_NRF51_UNIFIED -DMBED_TICKLESS -DDEVICE_INTERRUPTIN=1 -DTARGET_CORTEX -DDEVICE_I2C=1 -DDEVICE_PORTOUT=1 -DDEVICE_SERIAL=1 -DDEVICE_SPI_ASYNCH=1 -DTARGET_FF_ARDUINO -DDEVICE_PORTIN=1 -DTARGET_RELEASE -DTARGET_NORDIC -DDEVICE_SERIAL_FC=1 -DBLE_STACK_SUPPORT_REQD -DTARGET_MCU_NRF51_32K_UNIFIED -DTARGET_NRF51_DK -DDEVICE_SLEEP=1 -DTOOLCHAIN_GCC_ARM -DTARGET_MCU_NRF51822 -DTARGET_MCU_NORDIC_32K -DSOFTDEVICE_PRESENT -DNO_SYSTICK -DDEVICE_SPI=1 -DCMSIS_VECTAB_VIRTUAL_HEADER_FILE="cmsis_nvic.h" -DMBED_BUILD_TIMESTAMP=1525940526.38 -DDEVICE_ANALOGIN=1 -DDEVICE_PWMOUT=1 -DS130 -DTARGET_LIKE_CORTEX_M0 -DTARGET_MCU_NRF51_32K @F:\pi\lumos\angler\nrf51_test\BUILD\tests\NRF51_DK\GCC_ARM\.\mbed-os\TESTS\network\emac\.includes_8285364bd2c9f08b4f63dc09ac3f1a8b.txt -include F:\pi\lumos\angler\nrf51_test\BUILD\tests\NRF51_DK\GCC_ARM\.\mbed-os\TESTS\network\emac\mbed_config.h -MD -MF BUILD\tests\NRF51_DK\GCC_ARM\mbed-os\TESTS\network\emac\emac_ctp.d -o BUILD\tests\NRF51_DK\GCC_ARM\mbed-os\TESTS\network\emac\emac_ctp.o .\mbed-os\TESTS\network\emac\emac_ctp.cpp
        [DEBUG] Return: 1
        [DEBUG] Output: arm-none-eabi-g++: error: CreateProcess: No such file or directory
```
This is on a new, empty project with just mbed-os checked out.
I narrowed the problem down to 5038135c7d1302c2b62a8695c41de90a7e19617d (when #6577 was merged) .
`mbed compile ...` still works fine.
`mbed test -m lpc1768  -t GCC_ARM --verbose --clean` works fine.

If I checkout the previous commit 53aa1b04ab29617f20e20e3b191caeb82bbbd48a my original command for NRF51_DK works fine.
```
[DEBUG] Compile: F:\Program Files (x86)\GNU Tools ARM Embedded\6 2017-q1-update\bin\arm-none-eabi-g++ -std=gnu++98 -fno-rtti -Wvla -c -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -fmessage-length=0 -fno-exceptions -fno-builtin -ffunction-sections -fdata-sections -funsigned-char -MMD -fno-delete-null-pointer-checks -fomit-frame-pointer -Os -g1 -mcpu=cortex-m0 -mthumb -D__CORTEX_M0 -DNRF51 -D__MBED__=1 -DTARGET_LIKE_MBED -DTARGET_NRF51822 -DDEVICE_PORTINOUT=1 -D__MBED_CMSIS_RTOS_CM -DDEVICE_LOWPOWERTIMER=1 -DDEVICE_RTC=1 -DTOOLCHAIN_object -DDEVICE_SERIAL_ASYNCH=1 -D__CMSIS_RTOS -DTARGET_MCU_NRF51822_UNIFIED -DTOOLCHAIN_GCC -DDEVICE_I2C_ASYNCH=1 -DTARGET_CORTEX_M -DTARGET_SDK11 -DARM_MATH_CM0 -DDEVICE_SERIAL=1 -DTARGET_UVISOR_UNSUPPORTED -DTARGET_NRF5 -DFEATURE_BLE=1 -DTARGET_M0 -DTARGET_MCU_NRF51 -DCMSIS_VECTAB_VIRTUAL -DTARGET_MCU_NRF51_UNIFIED -DMBED_TICKLESS -DDEVICE_INTERRUPTIN=1 -DTARGET_CORTEX -DDEVICE_I2C=1 -DDEVICE_PORTOUT=1 -DMBED_BUILD_TIMESTAMP=1525940933.87 -DDEVICE_SPI_ASYNCH=1 -DTARGET_FF_ARDUINO -DDEVICE_PORTIN=1 -DTARGET_RELEASE -DTARGET_NORDIC -DDEVICE_SERIAL_FC=1 -DBLE_STACK_SUPPORT_REQD -DTARGET_MCU_NRF51_32K_UNIFIED -DTARGET_NRF51_DK -DDEVICE_SLEEP=1 -DTOOLCHAIN_GCC_ARM -DTARGET_MCU_NRF51822 -DTARGET_MCU_NORDIC_32K -DSOFTDEVICE_PRESENT -DNO_SYSTICK -DDEVICE_SPI=1 -DCMSIS_VECTAB_VIRTUAL_HEADER_FILE="cmsis_nvic.h" -DDEVICE_SPISLAVE=1 -DDEVICE_ANALOGIN=1 -DDEVICE_PWMOUT=1 -DS130 -DTARGET_LIKE_CORTEX_M0 -DTARGET_MCU_NRF51_32K @F:\pi\lumos\angler\nrf51_test\BUILD\tests\NRF51_DK\GCC_ARM\.includes_3f30d65de18a5bb42c23dcf47e9262b6.txt -include F:\pi\lumos\angler\nrf51_test\BUILD\tests\NRF51_DK\GCC_ARM\mbed_config.h -MD -MF BUILD\tests\NRF51_DK\GCC_ARM\mbed-os\drivers\SPISlave.d -o BUILD\tests\NRF51_DK\GCC_ARM\mbed-os\drivers\SPISlave.o F:\pi\lumos\angler\nrf51_test\mbed-os\drivers\SPISlave.cpp
[DEBUG] Return: 0
```

Go back to 5038135c7d1 (or anything newer) and it fails again.

This is all on Windows 10 x64 GCC_ARM.
Tested on versions 7-2017-q4-major and 6-2017-q1-update with no change.

### Issue request type

[ ] Question  
[ ] Enhancement  
[X] Bug

andrewleech, 2018-05-11:
Ok, on another test today, C files all compile just fine (arm-none-eabi-gcc), it's only c++ ones (arm-none-eabi-g++) that fail with the above error.

andrewleech, 2018-05-11:
**The failures appear to be due to command path length.**

To break it down further I picked two of the compile command from the library and the test sections of the compilation process and re-ran them manually with --verbose
library:
```
"F:\Program Files (x86)\GNU Tools ARM Embedded\6 2017-q1-update\bin\arm-none-eabi-g++" -std=gnu++98 -fno-rtti -Wvla -c -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -fmessage-length=0 -fno-exceptions -fno-builtin -ffunction-sections -fdata-sections -funsigned-char -MMD -fno-delete-null-pointer-checks -fomit-frame-pointer -Os -g1 -mcpu=cortex-m0 -mthumb -D__CORTEX_M0 -DNRF51 -D__MBED__=1 -DTARGET_LIKE_MBED -DTARGET_NRF51822 -DDEVICE_PORTINOUT=1 -D__MBED_CMSIS_RTOS_CM -DDEVICE_LOWPOWERTIMER=1 -DDEVICE_RTC=1 -DTOOLCHAIN_object -DDEVICE_SERIAL_ASYNCH=1 -D__CMSIS_RTOS -DTARGET_MCU_NRF51822_UNIFIED -DTOOLCHAIN_GCC -DDEVICE_I2C_ASYNCH=1 -DTARGET_CORTEX_M -DTARGET_SDK11 -DARM_MATH_CM0 -DTARGET_UVISOR_UNSUPPORTED -DTARGET_NRF5 -DFEATURE_BLE=1 -DTARGET_M0 -DTARGET_MCU_NRF51 -DCMSIS_VECTAB_VIRTUAL -DTARGET_MCU_NRF51_UNIFIED -DMBED_TICKLESS -DDEVICE_INTERRUPTIN=1 -DTARGET_CORTEX -DDEVICE_I2C=1 -DDEVICE_PORTOUT=1 -DDEVICE_SERIAL=1 -DDEVICE_SPI_ASYNCH=1 -DTARGET_FF_ARDUINO -DDEVICE_PORTIN=1 -DTARGET_RELEASE -DTARGET_NORDIC -DDEVICE_SERIAL_FC=1 -DBLE_STACK_SUPPORT_REQD -DTARGET_MCU_NRF51_32K_UNIFIED -DTARGET_NRF51_DK -DDEVICE_SLEEP=1 -DTOOLCHAIN_GCC_ARM -DTARGET_MCU_NRF51822 -DTARGET_MCU_NORDIC_32K -DSOFTDEVICE_PRESENT -DNO_SYSTICK -DDEVICE_SPI=1 -DCMSIS_VECTAB_VIRTUAL_HEADER_FILE="cmsis_nvic.h" -DDEVICE_SPISLAVE=1 -DDEVICE_ANALOGIN=1 -DDEVICE_PWMOUT=1 -DS130 -DMBED_BUILD_TIMESTAMP=1526003366.28 -DTARGET_LIKE_CORTEX_M0 -DTARGET_MCU_NRF51_32K @F:\pi\lumos\angler\nrf51_test\BUILD\tests\NRF51_DK\GCC_ARM\.includes_414f16aa25c6e0078eaf164d23209da7.txt -include F:\pi\lumos\angler\nrf51_test\BUILD\tests\NRF51_DK\GCC_ARM\mbed_config.h -MD -MF BUILD\tests\NRF51_DK\GCC_ARM\mbed-os\features\nanostack\mbed-mesh-api\source\CallbackHandler.d -o BUILD\tests\NRF51_DK\GCC_ARM\mbed-os\features\nanostack\mbed-mesh-api\source\CallbackHandler.o F:\pi\lumos\angler\nrf51_test\mbed-os\features\nanostack\mbed-mesh-api\source\CallbackHandler.cpp --verbose 2> library.log
```
test:
```
"F:\Program Files (x86)\GNU Tools ARM Embedded\6 2017-q1-update\bin\arm-none-eabi-g++"" -std=gnu++98 -fno-rtti -Wvla -c -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -fmessage-length=0 -fno-exceptions -fno-builtin -ffunction-sections -fdata-sections -funsigned-char -MMD -fno-delete-null-pointer-checks -fomit-frame-pointer -Os -g1 -mcpu=cortex-m0 -mthumb -D__CORTEX_M0 -DNRF51 -D__MBED__=1 -DTARGET_LIKE_MBED -DTARGET_NRF51822 -DDEVICE_PORTINOUT=1 -D__MBED_CMSIS_RTOS_CM -DDEVICE_LOWPOWERTIMER=1 -DDEVICE_RTC=1 -DTOOLCHAIN_object -DDEVICE_SERIAL_ASYNCH=1 -D__CMSIS_RTOS -DTARGET_MCU_NRF51822_UNIFIED -DTOOLCHAIN_GCC -DDEVICE_I2C_ASYNCH=1 -DTARGET_CORTEX_M -DMBED_BUILD_TIMESTAMP=1526003435.46 -DTARGET_SDK11 -DARM_MATH_CM0 -DTARGET_UVISOR_UNSUPPORTED -DTARGET_NRF5 -DFEATURE_BLE=1 -DTARGET_M0 -DTARGET_MCU_NRF51 -DCMSIS_VECTAB_VIRTUAL -DTARGET_MCU_NRF51_UNIFIED -DMBED_TICKLESS -DDEVICE_INTERRUPTIN=1 -DTARGET_CORTEX -DDEVICE_I2C=1 -DDEVICE_PORTOUT=1 -DDEVICE_SERIAL=1 -DDEVICE_SPI_ASYNCH=1 -DTARGET_FF_ARDUINO -DDEVICE_PORTIN=1 -DTARGET_RELEASE -DTARGET_NORDIC -DDEVICE_SERIAL_FC=1 -DBLE_STACK_SUPPORT_REQD -DTARGET_MCU_NRF51_32K_UNIFIED -DTARGET_NRF51_DK -DDEVICE_SLEEP=1 -DTOOLCHAIN_GCC_ARM -DTARGET_MCU_NRF51822 -DTARGET_MCU_NORDIC_32K -DSOFTDEVICE_PRESENT -DNO_SYSTICK -DDEVICE_SPI=1 -DCMSIS_VECTAB_VIRTUAL_HEADER_FILE="cmsis_nvic.h" -DDEVICE_SPISLAVE=1 -DDEVICE_ANALOGIN=1 -DDEVICE_PWMOUT=1 -DS130 -DTARGET_LIKE_CORTEX_M0 -DTARGET_MCU_NRF51_32K @F:\pi\lumos\angler\nrf51_test\BUILD\tests\NRF51_DK\GCC_ARM\.\mbed-os\features\frameworks\utest\TESTS\unit_tests\test_setup_failure\.includes_67223ea430597b1f82efe4193730bf51.txt -include F:\pi\lumos\angler\nrf51_test\BUILD\tests\NRF51_DK\GCC_ARM\.\mbed-os\features\frameworks\utest\TESTS\unit_tests\test_setup_failure\mbed_config.h -MD -MF BUILD\tests\NRF51_DK\GCC_ARM\mbed-os\features\frameworks\utest\TESTS\unit_tests\test_setup_failure\main.d -o BUILD\tests\NRF51_DK\GCC_ARM\mbed-os\features\frameworks\utest\TESTS\unit_tests\test_setup_failure\main.o .\mbed-os\features\frameworks\utest\TESTS\unit_tests\test_setup_failure\main.cpp --verbose 2> test.log
```

The outputs of these list the internal compiler command being used:
library:
```
 f:/program files (x86)/gnu tools arm embedded/6 2017-q1-update/bin/../lib/gcc/arm-none-eabi/6.3.1/cc1plus.exe -quiet -v -I F:/pi/lumos/angler/nrf51_test -I F:/pi/lumos/angler/nrf51_test/mbed-os -I F:/pi/lumos/angler/nrf51_test/mbed-os/cmsis -I F:/pi/lumos/angler/nrf51_test/mbed-os/cmsis/TARGET_CORTEX_M -I F:/pi/lumos/angler/nrf51_test/mbed-os/drivers -I F:/pi/lumos/angler/nrf51_test/mbed-os/events -I F:/pi/lumos/angler/nrf51_test/mbed-os/events/equeue -I F:/pi/lumos/angler/nrf51_test/mbed-os/features -I F:/pi/lumos/angler/nrf51_test/mbed-os/features/FEATURE_BLE -I F:/pi/lumos/angler/nrf51_test/mbed-os/features/FEATURE_BLE/ble -I F:/pi/lumos/angler/nrf51_test/mbed-os/features/FEATURE_BLE/ble/generic -I F:/pi/lumos/angler/nrf51_test/mbed-os/features/FEATURE_BLE/ble/pal -I F:/pi/lumos/angler/nrf51_test/mbed-os/features/FEATURE_BLE/ble/services -I F:/pi/lumos/angler/nrf51_test/mbed-os/features/FEATURE_BLE/source -I F:/pi/lumos/angler/nrf51_test/mbed-os/features/FEATURE_BLE/source/generic -I F:/pi/lumos/angler/nrf51_test/mbed-os/f [... cut ...]
If I run the above cc1plus command manually (in a batch file) I get the `The system cannot execute the specified program.` error.
If I then change the command to make all the project absolute paths relative, aka replace `F:/pi/lumos/angler/nrf51_test` with `.` it then runs completely fine.

To confirm all this I moved my test project from `F:\pi\lumos\angler\nrf51_test` to `F:\test` and the `mbed test` build worked fine.

**Can the build commands be updated to make all project paths relative?**
That should make it location agnostic.

SeppoTakalo, 2018-05-11:
This is similar to https://github.com/ARMmbed/mbed-os/issues/6335

Sadly, a problem we cannot fix. It is a Windows...

@0xc0170 @adbridge It is a time that we should start investigating workarounds for this.

andrewleech, 2018-05-11:
There are a lot of absolute paths used, mostly all the include paths listed in the autogenerated include file (eg. `BUILD\tests\NRF51_DK\GCC_ARM.includes_4d581b320aa0fc19b2c1af7426f743ac.txt`) 

The include paths here could all be made relative to the project dir instead which would solve the problem for most if not all projects currently... until a lot more include dirs are added.

adbridge, 2018-05-11:
@theotherjimmy Could we look at @andrewleech suggestions ?

theotherjimmy, 2018-05-11:
@adbridge Yeah! Just a sec.

TeroJaasko, 2018-05-14:
As a final resort, "-iprefix" and "-iwithprefix" options might also be used, eg. 
```-iprefix ./BUILD/tests/NRF51_DK/GCC_ARM/mbed-os/ -iwithprefix TARGET_CORTEX/rtx5/RTX/Source/TOOLCHAIN_GCC/TARGET_M0/```

kjbracey-arm, 2018-05-16:
Compilers generally have an option for "get files (or options)" from a file. For gcc it's `@file`.  Any reason we can't use that?

andrewleech, 2018-05-16:
As I mentioned earlier the @include.txt system is already in use, however the paths listed in the include file are all absolute paths. 

The include list file is passed to g++ by mbed which then unpacks it internally to call cc1plus.exe where it passes the full command line which is where it fails.

0xc0170, 2018-05-16:
Any progress for this one?

theotherjimmy, 2018-05-21:
Sorry, I was off last week. I'm picking up this investigation today.

0xc0170, 2018-06-13:
@andrewleech Thank for descriptive bug report. This is duplicate for #6335. I'll close this one, please monitor 6335.

0xc0170, 2018-06-19:
Reopening, similar to 6335 but this is building with `mbed compile` rather than using exporters.

theotherjimmy, 2018-06-19:
@0xc0170 This is `mbed test`, which is not the same as `mbed compile`

0xc0170, 2018-06-21:
@

closed 2018-06-21 by 0xc0170