ARMmbed/mbed-os #6868: RTC tests should start from year 2000+

MarceloSalazar, 2018-05-10:
### Description

While running RTC tests on a new platform that is coming, we realized that certain RTC tests are causing problems because are expecting the MCU to handle old years correctly, for example, 1970.

See:
https://github.com/ARMmbed/mbed-os/blob/master/TESTS/mbed_hal/rtc_time_conv/main.cpp
https://github.com/ARMmbed/mbed-os/blob/master/TESTS/mbed_hal/rtc_time/main.cpp#L58

These dates are not supported in all MCUs. They are not applicable anymore given we are in 2018.

This enhancement/bug fix request is to relax RTC tests to start from the year 2000 onwards.

[ ] Question  
[X] Enhancement  
[X] Bug  

@bulislaw @Przemek

MarceloSalazar, 2018-05-10:
Internal reference: IOTMORF-2349

MarceloSalazar, 2018-05-14:
Details in internal ticket.

0xc0170, 2018-05-17:
@mprse Please review

MarceloSalazar, 2018-05-23:
As per internal conversation with @bulislaw

The HW is handling years in BCD.

Our RTC spec is working in seconds from epoch and that's what they will get anyway. So in their implementation, they will need to handle the conversion from sec to their date in BCD, they can handle base manipulation internally. Very quick pseudocode with workaround.

```
// rtc_write(0); /* setting time to 1970-01-01 00:00:00 */
--
time_t base;
 
void rtc_write(time_t t)
{
base = first_addressable_date - t;
bcd_t bcd_data = seconds_to_bcd(t - base)
write_to_hw(bcd_data);
}
 
time_t rtc_read(void)
{
bcd_t bcd_data = read_from_hw();
time_t t = bcd_to_seconds(bcd_data);
 
return t + base;
}

```

@0xc0170 I'd suggest we set the appropriate labels and close this ticket. Thanks!

0xc0170, 2018-05-23:
Thanks for the update. I'll close this ticket as resolved, given the snippet above should help to fix the problem.

closed 2018-08-03 by cmonr