ARMmbed/mbed-os #6873: What is the ```uint32_t  _count;``` variable used for in Mutex Class?

daniel-cesarini-tridonic-com, 2018-05-11:
### Description

What is the ```uint32_t                  _count;``` variable used for?
https://github.com/ARMmbed/mbed-os/blob/30e39eeb10b2e0961068748651f001fe008b374a/rtos/Mutex.h#L156

It is being increased:
https://github.com/ARMmbed/mbed-os/blob/30e39eeb10b2e0961068748651f001fe008b374a/rtos/Mutex.cpp#L56
and decreased:
https://github.com/ARMmbed/mbed-os/blob/30e39eeb10b2e0961068748651f001fe008b374a/rtos/Mutex.cpp#L90
but I do not see any other usage.

### Issue request type

[X] Question  
[ ] Enhancement  
[ ] Bug  

FYI: @c1728p9 @MarceloSalazar @geky, @kjbracey-arm, @sg-

daniel-cesarini-tridonic-com, 2018-05-11:
FYI: @andreas-myka-tridonic-com, @markus-becker-tridonic-com

kjbracey-arm, 2018-05-11:
It's for debugging purposes only. Was introduced in #3648.

Only "use" is this in CondtionVariable.cpp:

    MBED_ASSERT(_mutex._count == 1);

daniel-cesarini-tridonic-com, 2018-05-11:
Should it then be "stripped" out in RELEASE build?

kjbracey-arm, 2018-05-11:
This was discussed in the review (with me on the "let's not have it there" side):

https://github.com/ARMmbed/mbed-os/pull/3648#discussion_r148197584

I did compensate for it going in by taking out another 16 bytes.

https://github.com/ARMmbed/mbed-os/pull/5428

There is a problem with adding/removing it for debug/release, and that's binary compatibility. The Mutex changes did cause issues with some binary drivers: https://github.com/ARMmbed/mbed-os/pull/5547

Don't think we can afford different structure layout for those core classes based on build type :(

kjbracey-arm, 2018-08-17:
I believe the question is answered.

closed 2018-08-17 by kjbracey-arm