ARMmbed/mbed-os #6877: Expose DMAUsage in I2C API

markus-becker-tridonic-com, 2018-05-11:
DMAUsage is not accessible in the current I2C API.

https://github.com/ARMmbed/mbed-os/blob/30e39eeb10b2e0961068748651f001fe008b374a/drivers/I2C.h#L190
https://github.com/ARMmbed/mbed-os/blob/master/drivers/I2C.cpp#L31

Neither a constructor nor a modifying function is available. Should this be exposed or are API users required to derive from I2C class (e.g. a DMAI2C class)?

@daniel-cesarini-tridonic-com @MarceloSalazar

0xc0170, 2018-05-17:
Hi,

this should be cleaned up (there will be I2c updates @bulislaw can provide more updates).

bulislaw, 2018-05-17:
We'll be cleaning up I2C HAL API in next release.

scartmell-arm, 2018-08-28:
Do you have a use case where you need to specify the `DMAUsage` parameter? The current plan would be to remove it altogether rather than expose it in the API.

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTHAL-77*

closed 2018-12-04 by ciarmcom