ARMmbed/mbed-os #6888: stm_spi_api.c gets stuck in infinite loop when hardware SS pin is used

michaelVisimid, 2018-05-11:
### Description

This is for all ST targets and tool chains. To reproduce initialize a SPI peripheral with a valid ssel pin then call the write function. The program will hand in an infinite loop in the write function call. For instance:

```

SPI spi(SPI_MOSI, SPI_MISO, SPI_SCK, SPI_CS);

int main() {
	spi.format(8, 0);
	spi.frequency(2000000);
	spi.write(0xAA);//infinite loop stm_spi_api.c line 407
	while (1) {
	}
}
```

The issue is fixed by adding the following below stm_spi_api.c line 165
```
    if (ssel != NC) {
        pinmap_pinout(ssel, PinMap_SPI_SSEL);
        handle->Init.NSS = SPI_NSS_HARD_OUTPUT;
    } else {
        handle->Init.NSS = SPI_NSS_SOFT;
    }
```
The hardware NSS line is meant more for a multi-master configuration than for addressing slave devices. ST forum discussion on the subject:

https://community.st.com/thread/39153-spi-master-nss-always-low-in-stm32f4

I would recommend the code be improved by always initializing the SPI peripheral to software NSS and using LL driver calls to toggle the NSS pin if one has been specified. This will make the SPI behavior match what the user expects when specifying a NSS pin.

### Issue request type

[ ] Question  
[X] Enhancement  
[X] Bug

0xc0170, 2018-05-14:
@ARMmbed/team-st-mcd Please review

bcostm, 2018-05-14:
I made a test with a NUCLEO_F303ZE board.

If I write `SPI spi(SPI_MOSI, SPI_MISO, SPI_SCK, SPI_CS);` there is a pinmap error at execution. Which is normal because SPI_CS is equal to PD_14 and there is no SPI1_SSEL function on this pin. Is it what you call "stuck" ?

If I write `SPI spi(SPI_MOSI, SPI_MISO, SPI_SCK, PA_4);` the code is working fine. Because the SPI_SSEL function is well available on this pin.

Anyway, we are agree with your fix to add `handle->Init.NSS = SPI_NSS_HARD_OUTPUT;` in the spi_init function. Is it possible for you to send a PR ?

Concerning your proposal:

> I would recommend the code be improved by always initializing the SPI peripheral to software NSS and using LL driver calls to toggle the NSS pin if one has been specified. This will make the SPI behavior match what the user expects when specifying a NSS pin.

We are not agree because the CS pin configuration must be done at application level and not in the SPI driver. Otherwise how will you manage different slave devices on the same SPI instance ? The SPI_CS in that case are DigitalOutput pins connected to all slave devices.

Do you agree ?

bcostm, 2018-05-15:
OK no problem. I will do it.

bcostm, 2018-05-22:
@michaelVisimid This issue is normally solved using PR 6951. Are you ok to close it ? Thanks

michaelVisimid, 2018-05-22:
Sounds good. Thank you for handling the PR.

closed 2018-05-22 by michaelVisimid