ARMmbed/mbed-os #6893: Remove FEATURE_COMMON_PAL from LoRaWAN stack

mattbrown015, 2018-05-14:
### Description
Remove use of FEATURE_COMMON_PAL from the LoRaWAN stack.

As I understand it, FEATURE_COMMON_PAL is unnecessary since [Get rid of FEATURE_COMMON_PAL and FEATURE_NANOSTACK #6577](https://github.com/ARMmbed/mbed-os/pull/6577).  But, I removed '"target.features_add": ["COMMON_PAL"]' from my mbed_app.json and all the LoRaWAN trace disappeared.

Just a tidy up really.

### Issue request type

[ ] Question  
[X ] Enhancement  
[ ] Bug

kivaisan, 2018-05-15:
Thanks. We'll check this.
Internal issue: IOTCELL-994

0xc0170, 2018-05-17:
This should fix it #6905

kivaisan, 2018-05-23:
Please note that I still have a PR pending for lorawan example application and I will merge it after mbed-os 5.8.5 is out.