ARMmbed/mbed-os #6894: Tidy mbed_trace.h Dependency on ns_types.h

mattbrown015, 2018-05-14:
### Description
`features/frameworks/mbed-trace/mbed-trace/mbed_trace.h` includes `features/frameworks/nanostack-libservice/mbed-client-libservice/ns_types.h` and it's not clear to me why 'mbed_trace' should be dependent on nanostack.

When I tried to exclude nanostack from my build the build failed because of this dependency.

I don't understand the significance of `YOTTA_CFG` in mbed_trace.h but hopefully it could just be removed and the inclusions be made simpler and more obvious.

Just a tidy up really.

### Issue request type

[ ] Question  
[X] Enhancement  
[ ] Bug

0xc0170, 2018-05-17:
@SeppoTakalo Please review

SeppoTakalo, 2018-05-17:
I don't see the problem?
You should not be touching the mbed build system by excluding folders.

Other than that, mbed Trace includes that header just for common types. Most probably can be replaced with `#include <stdint.h>` but not worth the effort, as it does not affect the build result.

mattbrown015, 2018-05-18:
Agreed only a small point so closing.

I 'touch the build system' using `.mbedignore`. Which, obviously, is a thing supported by the CLI, not something I invented.

I use  `.mbedignore` to exclude the features I'm not using because it is just a bit tedious watching and waiting while files I know I'm not using are compiled. Of course this doesn't happen every time you compile but when fiddling with a setting in the config that you don't understand it can happen quite a lot.

closed 2018-05-18 by mattbrown015