ARMmbed/mbed-os #6895: gerentea test_env defines NL and RCNL

drahnr, 2018-05-14:
[ ] Question  
[ ] Enhancement  
[ X] Bug  

```
/**
 *  Auxilary macros
 */
#define NL "\n"
#define RCNL "\r\n"
```
This can cause issues, since `NL` and `RCNL` are not uncommonly already defined in many projects.

0xc0170, 2018-05-15:
Did you face an error while implementing tests? I would not expect for these macros to be included to the application via regular include paths. How to reproduce this? 

Note: please leave headings (description, issue type) in the issue template.

0xc0170, 2018-05-15:
> No, a library actually implemented those. What I am asking for is either checking if they already exist and/or using namespaced constants instead.

This should be easily to fix. Can you send PR with a fix?

0xc0170, 2018-05-17:
@OPpuolitaival Please review

aashishc1988, 2018-09-25:
Apparently I do not have permissions to close this issue. Someone else would need to do that.

closed 2018-09-25 by drahnr