ARMmbed/mbed-os #6897: #defines for git info

bmcdonnell-ionx, 2018-05-14:
### Description
This feature was requested several years ago [on the Mbed forum](https://os.mbed.com/forum/bugs-suggestions/topic/3161/).

It would be useful to be able to access project Git hash / commit metadata from source code, probably as `#define`s, e.g. to output to console.

Example: https://stackoverflow.com/a/12368262

The primary thing to have is the commit hash. "Dirty" bool and tags would be nice to have. NB: I'm requesting metadata for the **project**, not for Mbed OS.

Hopefully you could make it work for Mbed CLI and exported projects.

### Issue request type
[ ] Question  
[X] Enhancement  
[ ] Bug

0xc0170, 2018-05-17:
Hi @bmcdonnell-ionx , I referenced this issue in the system stats API for consideration.

bmcdonnell-ionx, 2018-05-17:
@0xc0170, thanks for the heads-up.

If I understand correctly, #6821 is wanting (among other things) to possibly capture the Mbed OS SHA in the program? To be clear, that differs from this feature request, where I'm primarily looking to capture the user's project's SHA in the program. So they'd probably use a similar technique, but they are different features.

bmcdonnell-ionx, 2018-06-13:
This wouldn't necessarily have to be implemented as `#define`s, but the point is to make the info accessible from source code, and to have it automatically update as necessary on each build.

cmonr, 2018-08-03:
Closing since https://github.com/ARMmbed/mbed-os/pull/7434/ addresses this.

cmonr, 2018-08-03:
My mistake, misunderstood the issue.

bmcdonnell-ionx, 2018-08-03:
@cmonr,

> Closing since #7434 addresses this.

It does not. Please re-open.

I am asking for [1] git hash of [2] the user's project. By contrast, #7434 provides [1] version number (not hash) of [2] Mbed (not the user's project).

IDK if it would be part of `mbed-os` or `mbed-cli`.

theotherjimmy, 2018-08-03:
@bmcdonnell-ionx In the mean time you could run:
```bash
mbed compile -DPROJECT_HASH=$(git log -n1 --pretty='%H')
```

That'll probably force a rebuild for every new hash.

bmcdonnell-ionx, 2018-08-06:
@theotherjimmy, is the `-D` option to `mbed compile` documented somewhere? I can't find it.

And what about exporting the project to IDEs?

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCORE-396*

closed 2019-10-31 by ciarmcom