ARMmbed/mbed-os #6898: mbed-os/TESTS/netsocket/socket_sigio issue

jflynn129, 2018-05-14:
### inconsistent test failure
Failures when testing with socket_sigio running GreenTea netsocket test suit are being observed.  I have performed some preliminary investigatation and the problem is that the test routine 'get_data' is being called more than once and called after the socket has been closed.  From what I can see, the routine should be called once.
There are several EventQueues performing calls and it is possible the timing issues between them are causing failures.  It could also be possible that the tests that are stacked up back-to-back and one is trying to read a socket that another closed.  The ​Log data is:

**THE REQUEST IS SENT TO THE SERVER**
[1526319116.70][CONN][RXD] [WNC Driver]: ENTER socket_send() send 71 bytes
[1526319116.70][CONN][RXD] [WNC Driver]: ENTER tx_event(), socket 0
[1526319116.76][CONN][RXD] [WNC Driver]: EXIT tx_event, socket 0, data sent

**THE get_data routine is called in the test**
[1526319116.77][CONN][RXD] [WNC JMF: enter get_data()
[1526319116.77][CONN][RXD] Driver]: EXIT socket_send(), sent 71 bytes
[1526319116.77][CONN][RXD] [WNC Driver]: ENTER socket_recv(), socket 0, request 511 bytes
[1526319116.78][CONN][RXD] [WNC Driver]: ENTER rx_event() for socket 0
[1526319116.82][CONN][RXD] [WNC Driver]: EXIT rx_event(), socket 0, sechedule for more data.
[1526319116.84][CONN][RXD] [WNC Driver]: ENTER socket_recv(), socket 0, request 511 bytes
[1526319117.07][CONN][RXD] [WNC Driver]: ENTER rx_event() for socket 0
[1526319117.19][CONN][RXD] [WNC Driver]: data received on socket 0, cnt=389
[1526319117.19][CONN][RXD] [WNC Driver]: ENTER socket_recv(), socket 0, request 511 bytes
[1526319117.20][CONN][RXD] [WNC Driver]: EXIT

**data is received and returned, subsequently a call for an additional 122 bytes is made**
socket_recv(),socket 0, return 389 bytes
[1526319117.20][CONN][RXD] [WNC Driver]: ENTER socket_recv(), socket 0, request 122 bytes
[1526319117.21][CONN][RXD] [WNC Driver]: ENTER rx_event() for socket 0
[1526319117.25][CONN][RXD] [WNC Driver]: EXIT rx_event(), socket 0, sechedule for more data.
[1526319117.50][CONN][RXD] [WNC Driver]: ENTER rx_event() for socket 0
[1526319117.55][CONN][RXD] [WNC Driver]: EXIT rx_event(), socket 0, sechedule for more data.
[1526319117.80][CONN][RXD] [WNC Driver]: ENTER rx_event() for socket 0
[1526319117.84][CONN][RXD] [WNC Driver]: EXIT rx_event(), socket 0, sechedule for more data.
[1526319118.10][CONN][RXD] [WNC Driver]: ENTER rx_event() for socket 0
[1526319118.15][CONN][RXD] [WNC Driver]: EXIT rx_event(), socket 0, sechedule for more data.
[1526319118.41][CONN][RXD] [WNC Driver]: ENTER rx_event() for socket 0
[1526319118.45][CONN][RXD] [WNC Driver]: EXIT rx_event(), socket 0, sechedule for more data.
[1526319118.70][CONN][RXD] [WNC Driver]: ENTER rx_event() for socket 0
[1526319118.74][CONN][RXD] [WNC Driver]: EXIT rx_event(), socket 0, sechedule for more data.
[1526319118.99][CONN][RXD] [WNC Driver]: ENTER rx_event() for socket 0
[1526319119.04][CONN][RXD] [WNC Driver]: EXIT rx_event(), socket 0, sechedule for more data.
[1526319119.29][CONN][RXD] [WNC Driver]: ENTER rx_event() for socket 0
[1526319119.34][CONN][RXD] [WNC Driver]: EXIT rx_event(), socket 0, sechedule for more data.
[1526319119.59][CONN][RXD] [WNC Driver]: ENTER rx_event() for socket 0
[1526319119.64][CONN][RXD] [WNC Driver]: EXIT rx_event(), socket 0, sechedule for more data.
[1526319119.89][CONN][RXD] [WNC Driver]: ENTER rx_event() for socket 0
[1526319119.94][CONN][RXD] [WNC Driver]: EXIT rx_event(), socket 0, sechedule for more data.
[1526319120.19][CONN][RXD] [WNC Driver]: ENTER rx_event() for socket 0
[1526319120.31][CONN][RXD] [WNC Driver]: EXIT rx_event(), socket 0, sechedule for more data.
[1526319120.56][CONN][RXD] [WNC Driver]: ENTER rx_event() for socket 0
[1526319120.61][CONN][RXD] [WNC Driver]: EXIT rx_event(), socket 0, sechedule for more data.
[1526319120.86][CONN][RXD] [WNC Driver]: ENTER rx_event() for socket 0
[1526319120.90][CONN][RXD] [WNC Driver]: EXIT rx_event(), socket 0, sechedule for more data.
[1526319121.16][CONN][RXD] [WNC Driver]: ENTER rx_event() for socket 0
[1526319121.20][CONN][RXD] [WNC Driver]: EXIT rx_event(), socket 0, sechedule for more data.
[1526319121.46][CONN][RXD] [WNC Driver]: ENTER rx_event() for socket 0
[1526319121.50][CONN][RXD] [WNC Driver]: EXIT rx_event(), socket 0, sechedule for more data.
[1526319121.75][CONN][RXD] [WNC Driver]: ENTER rx_event() for socket 0
[1526319121.81][CONN][RXD] [WNC Driver]: EXIT rx_event(), socket 0, sechedule for more data.
[1526319122.05][CONN][RXD] [WNC Driver]: ENTER rx_event() for socket 0
[1526319122.10][CONN][RXD] [WNC Driver]: EXIT rx_event(), socket 0, sechedule for more data.
[1526319122.36][CONN][RXD] [WNC Driver]: ENTER rx_event() for socket 0
[1526319122.40][CONN][RXD] [WNC Driver]: EXIT rx_event(), socket 0, sechedule for more data.
[1526319122.65][CONN][RXD] [WNC Driver]: ENTER rx_event() for socket 0
[1526319122.70][CONN][RXD] [WNC Driver]: EXIT rx_event(), socket 0, sechedule for more data.
[1526319122.95][CONN][RXD] [WNC Driver]: ENTER rx_event() for socket 0
[1526319122.99][CONN][RXD] [WNC Driver]: EXIT rx_event(), socket 0, sechedule for more data.
[1526319123.25][CONN][RXD] [WNC Driver]: ENTER rx_event() for socket 0
[1526319123.29][CONN][RXD] [WNC Driver]: EXIT rx_event(), socket 0, sechedule for more data.
[1526319123.55][CONN][RXD] [WNC Driver]: ENTER rx_event() for socket 0
[1526319123.60][CONN][RXD] [WNC Driver]: EXIT rx_event(), socket 0, sechedule for more data.
[1526319123.86][CONN][RXD] [WNC Driver]: ENTER rx_event() for socket 0
[1526319123.90][CONN][RXD] [WNC Driver]: EXIT rx_event(), socket 0, sechedule for more data.
[... 70 lines cut ...]
[1526319127.01][CONN][INF] found KV pair in stream: {{__testcase_finish;Socket Attach Test;0;1}}, queued...
**SO THE TEST FAILS EVEN THOUGH IT PASSED**

I am able to make the test operate correctly by modifying the 'get_data' routine and putting the recvd.release() after the ASSERT test prior to the printf's effectively changing the timing as follows:

    TEST_ASSERT_EQUAL(result, true);
    // Signal that we have recvd
    recvd.release();
    printf("HTTP: Received %d chars from server\r\n", len);
    printf("HTTP: Received 200 OK status ... %s\r\n", found_200_ok ? "[OK]" : "[FAIL]");
    printf("HTTP: Received '%s' status ... %s\r\n", HTTP_HELLO_STR, found_hello ? "[OK]" : "[FAIL]");
    printf("HTTP: Received message:\r\n");
    printf("%s", buffer);
printf("JMF: done with get_data()\n");

so WRT the failure, it appears a timing issue exists in the test itself.   I don't understand the test sequence but it is not possible to read from a closed socket so timing is off somewhere.

This is occurring on a K64F running with an Avnet AT&T Started Kit IoT Shield.

### Issue request type
[ ] Question  
[ ] Enhancement  
[X] Bug  

@dlfryar

dlfryar, 2018-05-15:
cc: @bridadan

bridadan, 2018-05-15:
Thanks for filing the issue!

I took a look at this with @dlfryar and @geky and we drew a few conclusions:

1. We believe the problem is as described by @jflynn129. The test is using the EventQueue to move the sigio callback out of interrupt context, but sigio can be triggered for any reason at any time, so while the `get_data` callback is running, another callback is added to the queue. Therefore the test doesn't complete before the second callback is executed, and the test fails.
2. Really, an EventQueue isn't what's needed. We probably want a thread that's running, waiting on the sigio callback to release a semaphore, then do the `recv` calls in the thread until the HTTP response has been completely received. The thread can then exit and the test can complete. I think this applies to all of test cases inside the `socket_sigio` binary.

0xc0170, 2018-05-17:
@kjbracey-arm @mikaleppanen Please review

mikaleppanen, 2018-05-17:
Yes, there is an error as described above in the test case sigio handling.

closed 2018-08-10 by 0xc0170