ARMmbed/mbed-os #6900: how i can add pins PC14,PC15 in mbed os lib for target  DISCO-L072CZ-LRWAN1

helmi-haddad, 2018-05-15:


jeromecoutant, 2018-05-15:
Hi

PC14 and PC15 are connected to OSC32_IN and OUT

Can you explained more what is your request ?

Thx

0xc0170, 2018-05-15:
> No description provided.

Can you please use description to add information (already requested above) and use headline to introduce the issue you are creating? Please edit the issue.

helmi-haddad, 2018-05-15:
i have a specific hardware where i want to use the PC14 et PC15 ( I use that terget because i use the same STM32 chip ) / also , i can use the OSC_pins if i don't use the LSE .

jeromecoutant, 2018-05-15:
Add in PinNames.h file:
    PC_14 = 0x2E,
    PC_15 = 0x2F,
?

helmi-haddad, 2018-05-15:
how i can do it with mbed-os lib ?

bcostm, 2018-05-16:
> I use that terget because i use the same STM32 chip

Just to understand your application, do you use the CMWX1ZZABZ-091 Murata module or the STM32L072CZ MCU ?

helmi-haddad, 2018-05-16:
i use the STM32L072CZ MCU

bcostm, 2018-05-16:
OK. So, what you can do is to send a PR and add all the pins (in the PinNames.h file) which are available in the STM32L072CZ but not available in the Murata module. I would suggest also to add a comment for example:

```
// These pins are available on the STM32L072CZ only
PC_14 = 0x2E,
PC_15 = 0x2F,
...
```

The syntax is:
first nibble: 0=PA, 1=PB, 2=PC, etc...
second nibble: pin number

Can you do this ?

helmi-haddad, 2018-05-16:
no i can't because i dont know how , please can you do it ?

jeromecoutant, 2018-05-16:
I will do it.

@bcostm It is part of my current work of PeriphPin alignment

helmi-haddad, 2018-05-16:
@jeromecoutant and @bcostm thank you very much .

screamerbg, 2018-09-10:
Question considered resolved.

closed 2018-09-10 by screamerbg