ARMmbed/mbed-os #6902: UBLOX_C030_U201 bootloader/mbed-os target size is too large

JanneKiiskila, 2018-05-15:
### Description

Mbed OS target size with UBLOX_C030_U201 is much larger than with K64F.
It needs to be optimized for size.

```
git clone https://github.com/ARMmbed/mbed-bootloader
cd mbed-bootloader
mbed deploy
mbed compile -m UBLOX_C030_U201 -t GCC_ARM --profile=tiny.json
```
By default it uses Mbed OS 5.7.x but if you modify the `.mbedignore` file to include also these rows:
```
mbed-os/features/cellular/*
mbed-os/features/lorawan/*
mbed-os/features/nvstore/*
mbed-client-esfs/source/esfs_performance.c
```
you can then compile against Mbed OS 5.8.x and master.

Look at the output for the binary size, C030 is quite large compared for example to K64F.
The bootloader can't be larger than 32 kB in practice and now it's either over it or way too close  to it.
It needs to be optimized from size point of view.

K64F example size data with GCC_ARM.

```
+-----------------------------------------------+-------+-------+-------+
| Module                                        | .text | .data |  .bss |
+-----------------------------------------------+-------+-------+-------+
| [fill]                                        |    60 |     3 |  2519 |
| [lib]/c_nano.a                                |   740 |   100 |    12 |
| [lib]/gcc.a                                   |   832 |     0 |     0 |
| [lib]/misc                                    |   248 |     8 |    28 |
| fix-mbed-os-compile/pal_polyfill.o            |     4 |     0 |     0 |
| mbed-cloud-client/factory-configurator-client |  1688 |     0 |    92 |
| mbed-cloud-client/mbed-client-pal             |   720 |     0 |    56 |
| mbed-cloud-client/update-client-hub           |  2275 |     0 |    68 |
| mbed-os/drivers                               |  1322 |     0 |    20 |
| mbed-os/features                              |  2028 |     0 |     0 |
| mbed-os/hal                                   |  1339 |     4 |    66 |
| mbed-os/platform                              |   692 |     4 |    14 |
| mbed-os/targets                               |  9345 |    12 |   535 |
| minimal-printf/mbed_printf_implementation.o   |  1128 |     1 |   188 |
| minimal-printf/mbed_printf_wrapper.o          |    32 |     0 |     0 |
| sd-driver/SDBlockDevice.o                     |  3062 |     0 |     0 |
| source/active_application.o                   |   830 |     0 |    56 |
| source/bootloader_common.o                    |   246 |     4 | 16385 |
| source/bootloader_platform.o                  |    46 |     0 |     0 |
| source/main.o                                 |   783 |     4 |   480 |
| source/sotp_rot.o                             |    92 |     0 |     1 |
| source/upgrade.o                              |  1375 |     0 |     8 |
| Subtotals                                     | 28887 |   140 | 20528 |
+-----------------------------------------------+-------+-------+-------+
Total Static RAM memory (data + bss): 20668 bytes
Total Flash memory (text + data): 29027 bytes

Image: ./BUILD/K64F/GCC_ARM/mbed-bootloader.bin
```
vs. C030.

```
+-----------------------------------------------+-------+-------+-------+
| Module                                        | .text | .data |  .bss |
+-----------------------------------------------+-------+-------+-------+
| [fill]                                        |    54 |     3 |    22 |
| [lib]/c_nano.a                                |   740 |   100 |    12 |
| [lib]/gcc.a                                   |   832 |     0 |     0 |
| [lib]/misc                                    |   252 |    16 |    28 |
| fix-mbed-os-compile/pal_polyfill.o            |     4 |     0 |     0 |
| mbed-cloud-client/factory-configurator-client |  1212 |     0 |    76 |
| mbed-cloud-client/mbed-client-pal             |   648 |     4 |     4 |
| mbed-cloud-client/update-client-hub           |  2275 |     4 |    16 |
| mbed-os/drivers                               |  1442 |     4 |   120 |
| mbed-os/features                              |  1494 |     0 |     0 |
| mbed-os/hal                                   |  1431 |     4 |    66 |
| mbed-os/platform                              |   692 |     4 |    14 |
| mbed-os/targets                               | 15937 |     4 |   820 |
| minimal-printf/mbed_printf_implementation.o   |  1124 |     1 |    36 |
| minimal-printf/mbed_printf_wrapper.o          |    32 |     0 |     0 |
| sd-driver/SDBlockDevice.o                     |  3074 |     0 |     0 |
| source/active_application.o                   |   858 |     4 |     4 |
| source/bootloader_common.o                    |   246 |     4 | 16385 |
| source/bootloader_platform.o                  |    46 |     0 |     0 |
| source/main.o                                 |   779 |     8 |   328 |
| source/sotp_rot.o                             |    92 |     0 |     1 |
| source/upgrade.o                              |  1379 |     0 |     8 |
| Subtotals                                     | 34643 |   160 | 17940 |
+-----------------------------------------------+-------+-------+-------+
Total Static RAM memory (data + bss): 18100 bytes
Total Flash memory (text + data): 34803 bytes

Image: ./BUILD/UBLOX_C030_U201/GCC_ARM/mbed-bootloader.bin
```

The size difference is 5 kilobytes in that configuration, which is way too much.

[ ] Question  
[X] Enhancement  
[ ] Bug

JanneKiiskila, 2018-05-15:
@andreaslarssonublox  - is this something your team could look into together with ST @adustm ?

JanneKiiskila, 2018-05-15:
Internal issue; https://jira.arm.com/browse/IOTPART-5227

RobMeades, 2018-05-15:
Strange: this is a very standard STM32F4 target.  @JanneKiiskila: have you been able to build a boot loader with other STM32F4 targets (e.g. Odin W2), or does this just happen to be the first one?

JanneKiiskila, 2018-05-15:
Well, Nucleo F429ZI and Nucleo F411RE at least are much smaller. Here's F411RE as an example.

````
Building project mbed-bootloader-internal (NUCLEO_F411RE, GCC_ARM)
Scan: .
Scan: mbed
Scan: env
+---------------------------------------------+-------+-------+-------+
| Module                                      | .text | .data |  .bss |
+---------------------------------------------+-------+-------+-------+
| [fill]                                      |    53 |     6 |    17 |
| [lib]/c_nano.a                              |   740 |   100 |    12 |
| [lib]/gcc.a                                 |   832 |     0 |     0 |
| [lib]/misc                                  |   252 |    16 |    28 |
| fix-mbed-os-compile/pal_polyfill.o          |     4 |     0 |     0 |
| mbed-client-esfs/source                     |  1690 |     0 |    92 |
| mbed-client-pal/Source                      |   720 |     4 |     8 |
| mbed-os/drivers                             |  1514 |     4 |   120 |
| mbed-os/features                            |  2108 |     0 |     0 |
| mbed-os/hal                                 |  1411 |     4 |    68 |
| mbed-os/platform                            |   691 |     4 |    13 |
| mbed-os/targets                             |  9968 |     5 |   392 |
| minimal-printf/mbed_printf_implementation.o |  1124 |     1 |    76 |
| minimal-printf/mbed_printf_wrapper.o        |    32 |     0 |     0 |
| sd-driver/SDBlockDevice.o                   |  3092 |     0 |     0 |
| source/active_application.o                 |  1443 |     4 |     8 |
| source/bootloader_common.o                  |   246 |     4 | 16385 |
| source/bootloader_platform.o                |    46 |     0 |     0 |
| source/main.o                               |   779 |     8 |   336 |
| source/sotp_rot.o                           |    92 |     0 |     1 |
| source/upgrade.o                            |  1375 |     0 |     8 |
| update-client-hub/modules                   |  2275 |     4 |    20 |
| Subtotals                                   | 30487 |   164 | 17584 |
+---------------------------------------------+-------+-------+-------+
Total Static RAM memory (data + bss): 17748 bytes
Total Flash memory (text + data): 30651 bytes

Image: ./BUILD/NUCLEO_F411RE/GCC_ARM/mbed-bootloader-internal.bin
```

JanneKiiskila, 2018-05-15:
There is at least these files extra for UBLOX_C030_U201 (when looking at the `BUILD` -folder):

```
jankii01@ubuntu:~/mbed/mbed-bootloader-internal/BUILD/UBLOX_C030_U201/GCC_ARM/mbed-os/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F437xG/TARGET_UBLOX_C030$ ll
total 148
drwxrwxr-x 2 jankii01 jankii01  4096 May 15 11:34 ./
drwxrwxr-x 4 jankii01 jankii01  4096 May 14 12:24 ../
-rw-rw-r-- 1 jankii01 jankii01  6891 May 14 13:22 battery_charger_i2c.d
-rw-rw-r-- 1 jankii01 jankii01 23736 May 14 13:22 battery_charger_i2c.o
-rw-rw-r-- 1 jankii01 jankii01  6896 May 15 11:34 min_battery_voltage.d
-rw-rw-r-- 1 jankii01 jankii01 22468 May 15 11:34 min_battery_voltage.o
-rw-rw-r-- 1 jankii01 jankii01   279 May 15 11:34 onboard_modem_api.d
-rw-rw-r-- 1 jankii01 jankii01  1388 May 15 11:34 onboard_modem_api.o
-rw-rw-r-- 1 jankii01 jankii01  6393 May 15 11:34 PeripheralPins.d
-rw-rw-r-- 1 jankii01 jankii01 14472 May 15 11:34 PeripheralPins.o
-rw-rw-r-- 1 jankii01 jankii01  5982 May 15 11:34 system_clock.d
-rw-rw-r-- 1 jankii01 jankii01 11952 May 15 11:34 system_clock.o
-rw-rw-r-- 1 jankii01 jankii01  6854 May 15 11:34 ublox_low_level_api.d
-rw-rw-r-- 1 jankii01 jankii01 12800 May 15 11:34 ublox_low_level_api.o
```

RobMeades, 2018-05-15:
See #6011.  It fitted then...?

JanneKiiskila, 2018-05-15:
Updating the Mbed OS hash does help indeed.

0xc0170, 2018-05-15:
Thanks @RobMeades for the reference. This issue shall be resolved

JanneKiiskila, 2018-05-15:
Thanks @RobMeades  - can you still confirm it's OK to have the onboard_modem_api.c/h and min_battery_voltage.c/h -files in .mbedignore?

JanneKiiskila, 2018-05-16:
@RobMeades  - can we leave the modem API on the `.mbedignore` ?

In other words leave these in `.mbedignore`:
```
mbed-os/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F437xG/TARGET_UBLOX_C030/onboard_modem_api.c
mbed-os/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F437xG/TARGET_UBLOX_C030/onboard_modem_api.c
```

but not add these to `.mbedignore`.

```
mbed-os/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F437xG/TARGET_UBLOX_C030/min_battery_voltage.c	
mbed-os/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F437xG/TARGET_UBLOX_C030/min_battery_voltage.h
```

Personally,  I am starting to lean towards an idea that we should have some sort of "BOOTLOADER" config for the boards and with that you should be able to make a minimal image for bootloader that does the absolute minimum required for firmware update.  The applications you would compile with the full feature set and the bootloader with a strict, tight `BOOTLOADER` configuration.

RobMeades, 2018-05-16:
Yup, that's correct.  I think I tend to agree on the "BOOTLOADER" config: every board will be different and I no-one wants to lose flash space unnecessarily, so the opportunity (at least) to optimize makes sense, provided we aren't trying to achieve some nirvana where any application will work with any bootloader...?

RobMeades, 2018-05-16:
Just to be clear, leaving the `onboard_modem_api` out only works if you aren't planning to use cellular from the bootloader, which I guess is true.

JanneKiiskila, 2018-05-16:
Yep, there should not be any reason for bootloader to use cellular.

0xc0170, 2018-05-17:
Are there any actions that need to be taken @JanneKiiskila @RobMeades ? Or the update helps here as mentioned in the latest comment above?

JanneKiiskila, 2018-05-17:
This is OK to close (I will close). However, the bigger discussion of having a "BOOTLOADER" config for each board remains - I think we must do that and a product requirement [2777](https://jira.arm.com/browse/IOTPREQ-2777) has been filed.

closed 2018-05-17 by JanneKiiskila