ARMmbed/mbed-os #6909: REALTEK_RTL8195AM IPv6 build fails

juhaylinen, 2018-05-15:
### Description

Build fails if IPv6 support is enabled and target is REALTEK_RTL8195AM
```
Compile [ 96.7%]: timer_api.c
Compile [ 96.9%]: us_ticker.c
Compile [ 97.1%]: trng_api.c
Compile [ 97.2%]: rtw_emac.cpp
[Error] rtw_emac.cpp@54,12: 'ETHARP_HWADDR_LEN' was not declared in this scope
[ERROR] .\mbed-os\targets\TARGET_Realtek\TARGET_AMEBA\rtw_emac.cpp: In function 'uint8_t wlan_get_hwaddr_size(emac_interface_t*)':
.\mbed-os\targets\TARGET_Realtek\TARGET_AMEBA\rtw_emac.cpp:54:12: error: 'ETHARP_HWADDR_LEN' was not declared in this scope
     return ETHARP_HWADDR_LEN;

```

Target: REALTEK_RTL8195AM
Toolchain: GCC_ARM
mbed-cli 1.5.0
Mbed OS 5.8 and master (sha 80e109370)

Steps to reproduce:
For example, import mbed-os-example-wifi
```
mbed import mbed-os-example-wifi
cd mbed-os-example-wifi
```

Enable IPv6
```
"target_overrides": {
        "*": {
            "platform.stdio-convert-newlines": true,
            "lwip.ipv4-enabled": false,
            "lwip.ipv6-enabled": true
        },
```
Build
`mbed compile -t GCC_ARM -m REALTEK_RTL8195AM`

### Issue request type

[ ] Question  
[ ] Enhancement  
[X] Bug

0xc0170, 2018-05-15:
@ARMmbed/team-realtek please review

0xc0170, 2018-05-15:
Seems like the incorrect include or ? @juhaylinen Is this something we can easily fix?

juhaylinen, 2018-05-18:
Changed ETHARP_HWADDR_LEN to ETH_HWADDR_LEN and above issue seems fixed. But then build fails while compiling file mbed-os\targets\TARGET_Realtek\TARGET_AMEBA\sdk\common\network\dhcp\dhcps.c looks like IPv6 is not supported.

```
Compile [ 97.9%]: dhcps.c
[Warning] dhcps.c@120,23: implicit declaration of function 'ip4_addr4' [-Wimplicit-function-declaration]
[Warning] dhcps.c@379,75: implicit declaration of function 'ip4_addr1' [-Wimplicit-function-declaration]
[Warning] dhcps.c@379,104: implicit declaration of function 'ip4_addr2' [-Wimplicit-function-declaration]
[Warning] dhcps.c@379,133: implicit declaration of function 'ip4_addr3' [-Wimplicit-function-declaration]
[Warning] dhcps.c@380,2: implicit declaration of function 'IP4_ADDR' [-Wimplicit-function-declaration]
[Error] dhcps.c@690,38: 'struct netif' has no member named 'ip_addr'; did you mean 'ip6_addr'?
[Error] dhcps.c@692,35: 'struct netif' has no member named 'netmask'
[Error] dhcps.c@695,38: 'struct netif' has no member named 'gw'
[Error] dhcps.c@699,34: 'struct netif' has no member named 'ip_addr'; did you mean 'ip6_addr'?
[Error] dhcps.c@700,13: 'struct netif' has no member named 'netmask'
[Error] dhcps.c@703,14: 'struct netif' has no member named 'netmask'
[Warning] dhcps.c@705,43: passing argument 1 of 'lwip_htonl' makes integer from pointer without a cast [-Wint-conversion]
[Error] dhcps.c@705,28: assignment to expression with array type
[Warning] dhcps.c@706,41: passing argument 1 of 'lwip_htonl' makes integer from pointer without a cast [-Wint-conversion]
[Error] dhcps.c@706,27: assignment to expression with array type
[Warning] dhcps.c@707,39: passing argument 1 of 'lwip_htonl' makes integer from pointer without a cast [-Wint-conversion]
[Warning] dhcps.c@708,13: passing argument 1 of 'lwip_htonl' makes integer from pointer without a cast [-Wint-conversion]
[ERROR] .\mbed-os\targets\TARGET_Realtek\TARGET_AMEBA\sdk\common\network\dhcp\dhcps.c: In function 'save_client_addr':
.\mbed-os\targets\TARGET_Realtek\TARGET_AMEBA\sdk\common\network\dhcp\dhcps.c:120:23: warning: implicit declaration of function 'ip4_addr4' [-Wimplicit-function-declaration]
  uint8_t d = (uint8_t)ip4_addr4(client_ip);
...
```

M-ichae-l, 2018-05-21:
@juhaylinen 
The `dhcps.c` and `dhcps.h` should be deleted for MBED-OS. 
After I have fixed the compilation errors, the wifi connect will fail due to the DHCP_FAILURE. Are you expected the wifi connection error with ipv6 enabled? Currently, the DHCP only supports ipv4.

juhaylinen, 2018-05-21:
@M-ichae-l 
I tested quickly the changes you suggested. After successful build, the device gets an IPv6 address from our AP and tries to connect www.arm.com. The connection fails to NSAPI_ERROR_DNS_FAILURE but I think that's not related to this issue. Maybe www.arm.com is not IPv6 ready.

M-ichae-l, 2018-05-22:
@juhaylinen 
Thank you. Please refer to the #6974 for the fix.

cmonr, 2018-08-03:
Closing since fix has been merged.

closed 2018-08-03 by cmonr