ARMmbed/mbed-os #6916: NRF52: compiling master without SoftDevice

drahnr, 2018-05-15:
[X] Question  
[ ] Enhancement  
[ ] Bug  

Version: https://github.com/nello-io/mbed-os/#80e109370ddd3156126da447ca19852c53fcb90d

### Unclear how to disable

I followed 

https://github.com/ARMmbed/mbed-os/pull/5997/commits/9b98351b7ddc7659dfcc09bd3b78705a0000568e

Which resulted in this custom target:

```
{
    "QQQ": {
        "supported_form_factors": ["ARDUINO"],
        "inherits": ["MCU_NRF52832"],
        "macros_add" : ["CONFIG_NFCT_PINS_AS_GPIOS","MBED_TICKLESS"],
        "macros_remove" : ["BOARD_PCA10040"],
        "extra_labels_remove" : ["SOFTDEVICE_COMMON","SOFTDEVICE_S132"],
        "release_versions": ["2", "5"],
        "device_has_add": ["FLASH"],
        "device_name": "nRF52832_xxAA",
        "features_remove": [],
        "MERGE_BOOTLOADER": false,
        "MERGE_SOFTDEVICE": false
    },
```

but it complains about a missing `device.h` from `platform.h`

According to this the adjustments are non-trivial and I am not quite sure what exactly is worng in the above https://github.com/ARMmbed/mbed-os/pull/6826

Being stuck to master until 5.9 since I need features from 5.8 combined with NRF52 without Softdevice.

drahnr, 2018-05-16:
```
[Fatal Error] platform.h@29,10: device.h: No such file or directory
[ERROR] In file included from ./mbed-os/drivers/AnalogIn.h:19:0,
                 from ./mbed-os/drivers/AnalogIn.cpp:17:
./mbed-os/platform/platform.h:29:10: fatal error: device.h: No such file or directory
 #include "device.h"
          ^~~~~~~~~~
compilation terminated.
```

marcuschangarm, 2018-05-16:
You need to define your own `PinNames.h` and `device.h` for your target. Like this: https://github.com/ARMmbed/mbed-os/tree/master/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/TARGET_MCU_NRF52832/TARGET_NRF52_DK

It also looks like you are missing `"target.extra_labels_add": ["SOFTDEVICE_NONE"]` in your json file.

drahnr, 2018-05-17:
@marcuschangarm thanks! I was not aware of that. Since when has this been the case? Does my previous `inherit: NRF52_DK` solve this issue because `device.h` and `PinNames.h` are taken already?

marcuschangarm, 2018-05-17:
> Does my previous inherit: NRF52_DK solve this issue because device.h and PinNames.h are defined already?

Yes. I cleaned up `targets.json` and the inheritance in that file so that ideally people can inherit directly from MCU_NRF52832 or MCU_NRF52840 and not from the targets. That way, you can define only the pins that are available on your board that you care about.

Inheriting from MCU_NRF52x also means you are not forced into using these settings:
https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/mbed_lib.json#L87-L107

but you get the freedom to enable the erratas and set the debug flow control that apply to your target.

drahnr, 2018-06-04:
Also: The above also does not work for using it from within a `custom_targets.json` - see #7085

theotherjimmy, 2018-06-28:
@drahnr You may put it anywhere.

drahnr, 2018-06-28:
@theotherjimmy thanks for all the support, much appreciated!

closed 2018-06-28 by drahnr