ARMmbed/mbed-os #6920: Cannot compile with O0 optimization [nRF52-DK] [GCC-ARM]

tkstreet, 2018-05-15:
### Description

Target
NRF52

Toolchain: GCC-ARM (7 2017-q4-major)
Mbed SHA: 2104d8ab5 (HEAD) Merge pull request #6711 from marcuschangarm/cleanup-nrf5x
Issue: Will not compile with -O0 optimization.

After exporting my project from the mbed online compiler and changing the debug flags from -Os to -Og, I cannot compile mbed-os to produce the proper debugging output/symbols to debug.  If I compile with -O0 (as is recommended several places for debugging), the compile fails with:

>../mbed-os/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/sleep.c: In function 'hal_sleep':
../mbed-os/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/sleep.c:82:1: error: unrecognizable insn:
 }
 ^
(insn 16 15 17 2 (unspec_volatile [
            (mem/c:SI (plus:SI (reg/f:SI 105 virtual-stack-vars)
                    (const_int -4 [0xfffffffffffffffc])) [3 fpscr+0 S4 A32])
        ] VUNSPEC_SET_FPSCR) "../mbed-os/cmsis/TARGET_CORTEX_M/cmsis_gcc.h":734 -1
     (nil))
../mbed-os/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/sleep.c:82:1: internal compiler error: in extract_insn, at recog.c:2311

### Issue request type

[ ] Question  
[ ] Enhancement  
[X] Bug

pan-, 2018-05-16:
> Toolchain: GCC-ARM (7 2017-q4-major) 
We do not support gcc 7 yet; could you try with GCC 6 ?

0xc0170, 2018-05-17:
@tkstreet Can you test it with GCC 6 ?

I set -O0 for GCC, and built the latest master

```
Compile: arm-none-eabi-g++ -std=gnu++98 -fno-rtti -Wvla -c -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -fmessage-length=0 -fno-exceptions -fno-builtin -ffunction-sections -fdata-sections -funsigned-char -MMD -fno-delete-null-pointer-checks -fomit-frame-pointer -O0 -g1 -mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=softfp -DBOOTLOADER_ADDR=0x0 -DBOOTLOADER_SIZE=0x23000 -DAPPLICATION_ADDR=0x23000

.....

Build successes:
  * NRF52_DK::GCC_ARM::MBED-BUILD
  * NRF52_DK::GCC_ARM::TESTS-MBED_DRIVERS-TIMER
```

cc @marcuschangarm

mtfurlan, 2018-06-25:
I only see this on debug builds with  `7-2017-q4-major 7.2.1 20170904`, works fine with `6-2017-q2-update 6.3.1 20170620`

0xc0170, 2018-07-03:
Thanks @mtfurlan for confirming. As this is not currently supported toolchain, I'll close this issue.

We will address this once we update.

closed 2018-07-03 by 0xc0170