ARMmbed/mbed-os #6924: MBED_HEAP_STATS_ENABLED causes memory corrupt with ARM/ARMC6 in RTOS-less code

ccli8, 2018-05-16:
### Environment
**Target**
K64F

**mbed-os-example-blinky**
5.8.3

**mbed-os**
5.8.3

**ARM Toolchain**
5.24

**ARMC6 Toolchain**
6.10

### Steps to reproduce
1. Add `mbed_app.json` to enable MBED_HEAP_STATS_ENABLED
    ```
    {
        "target_overrides": {
            "K64F": {
                "target.macros_add": [
                    "MBED_HEAP_STATS_ENABLED=1",
                    "MBED_MEM_TRACING_ENABLED=1"
                ],
                "target.features_remove": ["LWIP"]
            }
        }
    }
    ```
1. Add `.mbedignore` for RTOS-less build
    ```
    mbed-os/rtos/*
    mbed-os/features/cellular/
    mbed-os/features/FEATURE_CLIENT/*
    mbed-os/features/FEATURE_COMMON_PAL/*
    mbed-os/features/FEATURE_UVISOR/*
    mbed-os/features/frameworks/*
    mbed-os/features/nanostack/*
    mbed-os/features/net/*
    mbed-os/features/netsocket/*
    mbed-os/features/nvstore/*
    mbed-os/features/storage/*
    mbed-os/events/*
    mbed-os/features/lorawan/*
    ```
1. Export to Keil project
1. Enter Keil project, compile it, and enter debug session
1. Before run, set breakpoint on the start of `malloc_wrapper`/`free_wrapper` in `mbed-os`/`platform`/`mbed_alloc_wrappers.cpp`.
1. Run the program, and you woud find `free_wrapper` is called before `malloc_wrapper` in C run-time initialization.

The memory hook mechanism relies on paired `malloc`/`free`, but it is broken here.

### Issue request type

[ ] Question  
[ ] Enhancement  
[X] Bug

0xc0170, 2018-05-17:
@deepikabhavnani Can you please review?

deepikabhavnani, 2018-07-05:
@ccli8 - Do we have this issue with mbed-os 5.9 and higher versions?

deepikabhavnani, 2018-07-19:
I was able to reproduce the issue and below is my observation. 
free_wrapper is called from __Heap_ProvideMemory function, in case of no RTOS it links to C standard library 'free` call, but in case of RTOS `_free_internal` call is used instead of standard C library call.

No RTOS
```
<P><STRONG><a name="[11e]"></a>__Heap_ProvideMemory</STRONG> (Thumb, 52 bytes, Stack size 0 bytes, C:\Keil_v5\ARM\ARMCC\Bin\..\lib\armlib\c_w.l(h1_extend.o)(.text))
<BR><BR>[Stack]<UL><LI>Max Depth = 608 + Unknown Stack Size
<LI>Call Chain = __Heap_ProvideMemory &rArr; free &rArr; free_wrapper &rArr; mbed_mem_trace_unlock &rArr; SingletonPtr&lang;PlatformMutex&rang;::operator -&rang;() &rArr; SingletonPtr&lang;PlatformMutex&rang;::get() &rArr; mbed_assert_internal &rArr; mbed_error_printf &rArr; mbed_error_vfprintf &rArr; serial_init &rArr;  mbed_assert_internal (Cycle)
</UL>
<BR>[Calls]<UL><LI><a href="#[115]">&gt;&gt;</a>&nbsp;&nbsp;&nbsp;free
</UL>
<BR>[Called By]<UL><LI><a href="#[e4]">&gt;&gt;</a>&nbsp;&nbsp;&nbsp;_init_alloc
<LI><a href="#[f8]">&gt;&gt;</a>&nbsp;&nbsp;&nbsp;__Heap_Full
</UL>
```
RTOS present
```
<P><STRONG><a name="[13a]"></a>__Heap_ProvideMemory</STRONG> (Thumb, 62 bytes, Stack size 8 bytes, C:\Keil_v5\ARM\ARMCC\Bin\..\lib\armlib\c_w.l(h1_extend_mt.o)(.text))
<BR><BR>[Stack]<UL><LI>Max Depth = 72<LI>Call Chain = __Heap_ProvideMemory &rArr; _free_internal &rArr; _mutex_release &rArr; osMutexRelease &rArr; EvrRtxMutexError &rArr;  mbed_error (Cycle)
</UL>
<BR>[Calls]<UL><LI><a href="#[112]">&gt;&gt;</a>&nbsp;&nbsp;&nbsp;_free_internal
</UL>
<BR>[Called By]<UL><LI><a href="#[fb]">&gt;&gt;</a>&nbsp;&nbsp;&nbsp;_init_alloc
<LI><a href="#[10f]">&gt;&gt;</a>&nbsp;&nbsp;&nbsp;__Heap_Full
</UL>
```

No RTOS: heap stats disabled
```
<P><STRONG><a name="[11c]"></a>__Heap_ProvideMemory</STRONG> (Thumb, 52 bytes, Stack size 0 bytes, C:\Keil_v5\ARM\ARMCC\Bin\..\lib\armlib\c_w.l(h1_extend.o)(.text))
<BR><BR>[Stack]<UL><LI>Max Depth = 16<LI>Call Chain = __Heap_ProvideMemory &rArr; free
</UL>
<BR>[Calls]<UL><LI><a href="#[f9]">&gt;&gt;</a>&nbsp;&nbsp;&nbsp;free
</UL>
<BR>[Called By]<UL><LI><a href="#[e4]">&gt;&gt;</a>&nbsp;&nbsp;&nbsp;_init_alloc
<LI><a href="#[f8]">&gt;&gt;</a>&nbsp;&nbsp;&nbsp;__Heap_Full
</UL>
```

Not sure what influences compiler decision to pick `free` vs `_free_internal`. Need to dig more

deepikabhavnani, 2018-07-24:
Hello All,

I got response from compiler team regarding call to `free` vs `_free_internal`. The compiler selects a thread safe version (which uses _free_internal) if the symbol ‘_mutex_acquire’ is present, which is the case of RTOS version.
`free` is a wrapper around `_free_internal` in case the thread safe version is selected.

__Heap_ProvideMemory is given some memory block which it tries to put into the heap data structure (free list) so that future malloc calls can use it. This data structure initialization is already done by _free_internal so instead of duplicating it, this function is used for both a regular free matching a malloc call, and an initial (non-matching) ‘free’ to initialize the memory for use as heap. 

Hence we see call to `free` without `malloc` in case of no RTOS.

deepikabhavnani, 2018-07-24:
Closing this issue, as we can't do anything; it is compilers decision to call `free` vs `free_internal`.

closed 2018-07-24 by deepikabhavnani

ccli8, 2018-07-25:
@deepikabhavnani So we can conclude that `MBED_HEAP_STATS_ENABLED` cannot be enabled in rtos-less code with ARM/ARMC6?

deepikabhavnani, 2018-07-25:
It can be enabled with some synchronisation to guard ‘MBED_HEAP_STATS_ENABLED’ function in free_wrapper till first call to ‘malloc’.

deepikabhavnani, 2018-09-10:
@ccli8 - Can you check if https://github.com/ARMmbed/mbed-os/pull/8062 will work for extra un-accounted free in ARM/ARMC6 as well for rtos-less code.

ccli8, 2018-09-11:
@deepikabhavnani Checking ARM/ARMC6 for rtos-less code, #8062 can fix the memory corrupt issue. I have one recommendation: it would decrease probability of mis-judge if signature can be cleared in `free`.