ARMmbed/mbed-os #6934: EventQueue Debugging Guidance

pauluap, 2018-05-17:
### Description

mbed-os 5.8
C++11

I'm running into a hard fault issue.  Producing a crash dump suggests that the exception occurs at this line - https://github.com/ARMmbed/mbed-os/blob/master/events/equeue/equeue.c#L264

I've stared at the code awhile and am still not quite sure what's going on.  This question is not a request to solve this particular problem, but rather a general question for debugging `EventQueue`s

I have multiple event queues in my application so setting a breakpoint in the equeue code isn't going to help.  Is there any techniques that I can use to zero into a specific event queue dispatch run?  I suppose I could find a memory location for an element within the equeue struct to set a data watchpoint on.  If there's nicer techniques that you can share, that'd be awesome.

An approximation of the segment of code that I think is the location of the problem is

```
struct A {
  bool thing;
};

struct B {
  uint16_t thing;
};

struct User {
  EventQueue queue;

  void Run() { queue.dispatch_forever(); }

  template <typename Stuff>
  void CallDoStuff(const Stuff stuff) {
    queue.call(this, static_cast<void(User::*)(const Stuff)>(&User::DoStuff), stuff);
  }

  void DoStuff(const A a) { std::cout << "A!\n"; }
  void DoStuff(const B b) { std::cout << "B!\n"; }
};

User user;
Thread thread;

void main()
{
  thread.start(Callback<void()>(&user, User::Run));
  user.CallDoStuff(A{true});
  user.CallDoStuff(B{0xC0DE});
}
```

While I didnt try to ensure that this snippet compiles, my actual code does compile with a number of analogoues to the `DoStuff` overloads.  I'm wondering if the code got generated incorrectly, but looking at the elf file, I don't see anything jumping out.  The equeue could have overflowed, but that's not an abnormal condition, and I don't expect it to lead to a hard fault.

I did get a core dump using https://github.com/adamgreen/CrashCatcher but I'm embarrassingly new to post-mortem investigation with gdb and core dumps so I would like to be able to catch the issue in the act so to speak.

So, my question:  Is there any techniques or code changes I can do to:
- Examine the dispatch of a specific queue
- Examine the processing of a specific event (possibly periodic)

### Issue request type

[X] Question  
[ ] Enhancement  
[ ] Bug

0xc0170, 2018-05-17:
@geky @kjbracey-arm @SenRamakri Please review

adamgreen, 2018-06-14:
@pauluap If you still need help, let me know and I can work with you to help debug your issue.

pauluap, 2018-06-14:
Thanks.  I think that the underlying question is pretty broad, how to debug different threads running through the same code.  If there's any good existing guidance, I'd appreciate any links or pointers.

Specifically,  I found that the problem I had been experiencing seems to be a combination of the event queue not having enough memory and repetition of `call()` after queue exhaustion.

I solved my particular problem by increasing memory and grouping the sequential calls into one call that passes a struct of data.

I figured I'd wait till the 5.9 release furor died down then go back and try to narrow down the triggering issue.

I'll open a new issue once I've done that.

adamgreen, 2018-06-14:
I am not sure how to answer your broader question. However, if you start debugging a similar issue again in the future and want help on how to proceed on a particular debug session, feel free to ping me and I will do my best to point you in the right direction for that particular session.

closed 2018-06-14 by pauluap