ARMmbed/mbed-os #6954: EV_COG_AD4050LZ mbed projects fails to build with CrossCore Embedded Studio

andrew-mclachlan, 2018-05-18:
### Description

LD scripts for EV_COG_AD4050LZ (and 3029) require preprocessing but this does not work for all toolchains.

Steps to reproduce:
pip install mbed-cli
mbed import mbed-os-example-blinky
cd mbed-os-example-blinky
mbed export -i cces -m EV_COG_AD4050LZ
Open CCES, import mbed-os-example-blinky/cces project.
Build
First build error is because `softfp` is not found. Remove it from Libraries.
Build

Building target: mbed-os-example-blinky
Invoking: CrossCore GCC ARM Embedded C++ Linker
arm-none-eabi-g++ -T/home/amclach/mbed-os-example-blinky/cces/../mbed-os/targets/TARGET_Analog_Devices/TARGET_ADUCM4X50/TARGET_ADUCM4050/TOOLCHAIN_GCC_ARM/ADuCM4050.ld -Wl,--gc-sections -Wl,--wrap,main -Wl,--wrap,_malloc_r -Wl,--wrap,_free_r -Wl,--wrap,_realloc_r -Wl,--wrap,_memalign_r -Wl,--wrap,_calloc_r -Wl,--wrap,exit -Wl,--wrap,atexit -Wl,-n -mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=softfp -lstdc++ -lsupc++ -lm -lc -lgcc -lnosys -mcpu=cortex-m4 -mthumb -specs=nosys.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -o  "mbed-os-example-blinky" @input-file.txt -lm
makefile:128: recipe for target 'mbed-os-example-blinky' failed
/opt/analog/cces/2.8.0/ARM/gcc-arm-embedded/bin/../lib/gcc/arm-none-eabi/6.3.1/../../../../arm-none-eabi/bin/ld:/home/amclach/mbed-os-example-blinky/cces/../mbed-os/targets/TARGET_Analog_Devices/TARGET_ADUCM4X50/TARGET_ADUCM4050/TOOLCHAIN_GCC_ARM/ADuCM4050.ld:8: ignoring invalid character `#' in expression
/opt/analog/cces/2.8.0/ARM/gcc-arm-embedded/bin/../lib/gcc/arm-none-eabi/6.3.1/../../../../arm-none-eabi/bin/ld:/home/amclach/mbed-os-example-blinky/cces/../mbed-os/targets/TARGET_Analog_Devices/TARGET_ADUCM4X50/TARGET_ADUCM4050/TOOLCHAIN_GCC_ARM/ADuCM4050.ld:8: syntax error
collect2: error: ld returned 1 exit status
make: *** [mbed-os-example-blinky] Error 1

#defines in LD scripts are not pre-processed and a separate tool is not run.

If you export to a Makefile `mbed export -i make_gcc_arm -m EV_COG_AD4050LZ`, then you'll notice a pre-processor step:

$(PROJECT).link_script.ld: $(LINKER_SCRIPT)
        @$(PREPROC) $< -o $@

### Issue request type
    [X ] Bug

0xc0170, 2018-05-18:
@ARMmbed/team-analogdevices Please review

Roshandcuk, 2018-05-21:
Hi, what steps do you take to;

Remove [softfp] from Libraries?

Thanks,

andrew-mclachlan, 2018-05-21:
Hi @Roshandcuk 

Sadly, the newly created CCES project ends up with an entry 'softfp' in `<option id="arm.linker.option.additionaloptions.348996874" name="Additional options for the linker:" superClass="arm.linker.option.additionaloptions" valueType="stringList">`. You can quickly edit the cces/.cproject file and remove that entry to keep you moving foward with the project.

```
<listOptionValue builtIn="false" value="-lnosys"/>
<listOptionValue builtIn="false" value="softfp"/>
```

However, even after resolving that particilar issue, the LD script(s) for the EV_COG_AD3029/4050 require pre-processing. Something that the GCC linker does not support and an mbed exported Makefile handles by adding an additional step.

You can a) modify the LD script in `./mbed-os/targets/TARGET_Analog_Devices/TARGET_ADUCM302X/TARGET_ADUCM3029/TOOLCHAIN_GCC_ARM/ADuCM3029.ld` b) add a pre-build step to your newly created CCES project and run the LD script through the pre-processor, and add an entry to -T options or b) copy `./mbed-os/targets/TARGET_Analog_Devices/TARGET_ADUCM302X/TARGET_ADUCM3029/TOOLCHAIN_GCC_ARM/ADuCM3029.ld` into your project, modify it and add a -T option.

andrew-mclachlan, 2018-05-25:
This issue requires an update to CrossCore Embedded Studio to pre-process the LD script.

0xc0170, 2018-05-30:
> This issue requires an update to CrossCore Embedded Studio to pre-process the LD script.

Does this resolves this issue? should this be noted in exporter documentation?

0xc0170, 2018-05-30:
I read the long discussion in 6690 PR but do not see there the solution to this issue

andrew-mclachlan, 2018-05-30:
Hi @0xc0170 -
It is my understanding that CrossCore Embedded Studio (CCES) needs to handle pre-processing of LD scripts in order to support an ARM mbed project. I see that the GNU ARM Eclipse plugins pre-process the LD script. Therefore an update is needed to CCES in order for the exported Analog Devices COG mbed project to build with CCES.

andrew-mclachlan, 2018-07-18:
Confirming that this Issue has been fixed for CrossCore Embedded Studio (CCES) version 2.9.0.

closed 2018-07-18 by 0xc0170