ARMmbed/mbed-os #6966: ARM_CM3DS_MPS2+ reboot.txt doesn't reboot the device

studavekar, 2018-05-21:
### Description
- Type: Bug 
- Priority: Major 

- also see : https://github.com/ARMmbed/mbed-os/issues/6460#issuecomment-389209451 
---------------------------------------------------------------
## Bug
**Target**
ARM_CM3DS_MPS2

**Toolchain:**
GCC_ARM|ARM|IAR

**mbed-cli version:** 
1.5.0

**mbed-os sha:**
ea13262 Merge pull request #6951 from bcostm/fix_spi_nss_config

**Expected behavior**
During the execution of tests, the device should always reboot using reboot.txt

**Actual behavior**
Device doesn't reboot even when a reboot.txt file is created.

**Steps to reproduce**
Running tests in loop, any toolchain should be file
mbed test -m ARM_CM3DS_MPS2 -t GCC_ARM -v

[ ] Question  
[ ] Enhancement  
[X ] Bug  

This is a blocker for adding device to CI .

cc : @ashok-rao @hug-dev

0xc0170, 2018-05-22:
cc @abhishek-pandit

0xc0170, 2018-05-30:
We created an internal ticket for this to track, so far no progress.

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTPART-6063*

studavekar, 2019-05-23:
closing it.

closed 2019-05-23 by studavekar