ARMmbed/mbed-os #6975: GR-PEACH/LYCHEE: mbed_blinky (Mbed 2) fails to compile

toyowata, 2018-05-22:
## Bug

**Target**
GR-PEACH and GR-LYCHEE

**Toolchain:**
Online compiler

**Expected behavior**
mbed_blinky compiles successfully and .bin file downloads for GR-PEACH/LYCHEE target.

**Actual behavior**
I got following errors:
```
Error: "/tmp/BWGgYX", line 9: Error:  #5: cannot open source input file "mem_RZ_A1LU.h": No such file or directory
Error:   #include "mem_RZ_A1LU.h"
Error:                           ^
Error: /tmp/BWGgYX: 0 warnings, 1 error
Error: Pre-processor step failed for '/extras/mbed/TARGET_GR_LYCHEE/TOOLCHAIN_ARM_STD/MBRZA1LU.sct'
Error: Image needs at least one load region.
Error: Not enough information to list the image map.
Info: Unable to download. Fix the reported errors...
```

**Steps to reproduce**
* Import https://os.mbed.com/teams/mbed/code/mbed_blinky/
* Update mbed library to latest version
* Set target to Renesas GR-PEACH or GR-LYCHEE

**Other information**

The similar report here https://github.com/ARMmbed/mbed-os/issues/6436 is already fixed (Mbed OS 5).

0xc0170, 2018-05-22:
@ARMmbed/mbed-os-tools @ARMmbed/team-renesas

toyowata, 2018-05-24:
Internal reference: IOTMORF-2375

0xc0170, 2018-08-27:
@toyowata @ARMmbed/team-renesas Is this still an issue ? I am at the moment not able to load compiler, will test

TomoYamanaka, 2018-08-28:
I checked this issue by using mbed lib 169, unfortunately the above failure still occurs.
The Online compiler's message is here.
```
Error: "/tmp/0OOcxS", line 9: Error:  #5: cannot open source input file "mem_RZ_A1H.h": No such file or directory
```
I think that this issue is unexpected behavior by Online Compiler.

@ARMmbed/mbed-os-tools could you confirm this behavior?

toyowata, 2018-09-28:
# Step to reproduce by offline build tool

```
hg clone https://MACRUM@os.mbed.com/teams/mbed/code/mbed_blinky/
cd mbed_blinky
mbed deploy
mbed compile -m GR_LYCHEE -t arm
```
# Build error log

```
[mbed] WARNING: Could not find mbed program in current path "C:\test\mbed2\mbed_blinky".
       You can fix this by calling "mbed new ." in the root of your program.
---
Using targets from C:\test\mbed2\mbed_blinky\mbed\e95d10626187\targets.json
Building project mbed_blinky (GR_LYCHEE, ARM)
Scan: mbed_blinky
Compile [100.0%]: main.cpp
Link: mbed_blinky
"C:\Users\toywat01\AppData\Local\Temp\p624-2", line 9: Error:  #5: cannot open source input file "mem_RZ_A1LU.h": No such file or directory
  #include "mem_RZ_A1LU.h"
                          ^
C:\Users\toywat01\AppData\Local\Temp\p624-2: 0 warnings, 1 error
Error: L6636E: Pre-processor step failed for './BUILD/GR_LYCHEE/ARM/.link_script.sct'
Error: L6372E: Image needs at least one load region.
Not enough information to list the image map.
Finished: 1 information, 0 warning and 2 error messages.
[ERROR] "C:\Users\toywat01\AppData\Local\Temp\p624-2", line 9: Error:  #5: cannot open source input file "mem_RZ_A1LU.h": No such file or directory
  #include "mem_RZ_A1LU.h"
                          ^
C:\Users\toywat01\AppData\Local\Temp\p624-2: 0 warnings, 1 error
Error: L6636E: Pre-processor step failed for './BUILD/GR_LYCHEE/ARM/.link_script.sct'
Error: L6372E: Image needs at least one load region.
Not enough information to list the image map.
Finished: 1 information, 0 warning and 2 error messages.

[mbed] ERROR: "c:\python27\python.exe" returned error.
       Code: 1
       Path: "C:\test\mbed2\mbed_blinky"
       Command: "c:\python27\python.exe -u C:\test\mbed2\mbed_blinky\.temp\tools\make.py -t arm -m GR_LYCHEE --source . --build .\BUILD\GR_LYCHEE\ARM"
       Tip: You could retry the last command with "-v" flag for verbose output
---
```

It seems that the linker script couldn't find the `mem_RZ_A1LU.h` include file.  The header file copied to the path below, but not in the same path of the linker script.

```
mbed_blinky\mbed\e95d10626187\TARGET_GR_LYCHEE\TARGET_RENESAS\TARGET_RZ_A1XX\TARGET_GR_LYCHEE\device\TOOLCHAIN_ARM_STD\mem_RZ_A1LU.h
```

I locally modified the script file as attached (remove inclusion form the script file and add macros) and I didn't get the build error.

[MBRZA1LU.zip](https://github.com/ARMmbed/mbed-os/files/2426363/MBRZA1LU.zip)

@ARMmbed/mbed-os-tools 
I don't know why we see the difference behavior between Mbed 2 and Mbed OS 5 by ARM toolchain build.  Any comment on this?

@ARMmbed/team-renesas  @TomoYamanaka 
Can you please review and test this?

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCORE-109*

TomoYamanaka, 2018-10-11:
@toyowata 
Sorry for this late reply.

> @ARMmbed/team-renesas @TomoYamanaka
> Can you please review and test this?

Your revised codes look good to me, test was also successful.

TomoYamanaka, 2018-10-30:
@toyowata 
Could you check this issue after Mbed 2(Classic) library updates because PR #8374 was merged into Mbed OS 5.10.2 that is latest release package?

rezaei99, 2019-01-11:
I face something similar. When I want to build "mbes-os-example-blinky" after exporting from mbed-online-compiler on Keil uvision5, I face this error :
`.\BUILD\mbed-os-example-blinky.axf:error: L6031U: Could not open scatter description file /filer/web_data/repo_dicts/github.com/armmbed/mbed-os/c966348d3f9ca80843be7cdc9b748f06ea73ced0/targets/TARGET_NXP/TARGET_LPC176X/device/TOOLCHAIN_ARM_STD/LPC1768.sct: No such file or directory` 

for local usage it should be :
`\mbed-os-example-blinky\mbed-os\targets\TARGET_NXP\TARGET_LPC176X\device\TOOLCHAIN_ARM_STD\LPC1768.sct`

SenRamakri, 2019-01-24:
@toyowata - I tried compiling blinky(with online compiler) with version 5.11.1 and compiled successfully. Is this issue still valid?
@rezaei99 - Your issue may be could be different as you are also exporting.

SakethKulapaka, 2019-01-25:
I just got my hands on the GR-PEACH full board yesterday. 
@toyowata been getting the same error on the online compiler.

SakethKulapaka, 2019-01-25:
I created a new program and chose the Blinky LED Hello World template. Except for a warning, it runs perfectly.

toyowata, 2019-01-27:
@SakethKulapaka 
I still got an error below:

[image]
[image]

fullbeat, 2019-05-18:
This same error is still present. "Blinky LED Hello World" will not build. I get the exact same error as @toyowata 

@SakethKulapaka, could you share screenshots of your project and the warning messages?

fullbeat, 2019-05-18:
Older blinky projects pull in a full set of mbed scr files. They appear to be missing from the newer reference project, even with or without choosing "Update the program and libraries to latest revision"

[image]

toyowata, 2019-06-11:
The issue is closed as we have an workaround for this i.e.  migrate to Mbed OS 5.

closed 2019-06-11 by toyowata